ETLCPP/etl #421: Default empty function for delegate instead of nullptr

legier, 2021-08-31:
It would be nice to have a switch, where the developer can decide, whether in case of an uninitialized delegate it should throw an assert/call nullptr, or just call an empty default function.

jwellbelove, 2021-09-01:
The only way to implement that would be to return a default constructed `TReturn` type, which may not be appropriate for all return types.
I'm not sure that 'silently failing' is a good idea. If you think that a delegate my be invalid then checking `is_valid()` before the call may be more appropriate.

legier, 2021-09-01:
Nobody says that calling empty delegate is failing. In case of event handler it just means, that nobody signed up for this event.
On the same hand, here: (https://github.com/ETLCPP/etl/issues/145) you rejected internal validity check, due to performance reasons, and here you propose to check it externally.

jwellbelove, 2021-09-01:
The validity check is optional (i.e. You don't pay for what you don't use). If your code guarantees that the delegate will never be invalid then a check (internal or external) is redundant. On the other hand, if your delegates may, or may not be, valid, then an additional external validity check should be required.

I'm open to the idea of adding another member function, something like `call_if_valid()` and even `call_or(<function>)`.

jwellbelove, 2021-09-02:
I am experimenting with adding the following to `etl:delegate`

**call_if**
```
etl::optional<TReturn> call_if(TParams... args) const
    Calls the delegate if valid.
    The return value is valid if the delegate was called.
```

**call_if**
```
bool call_if(TParams... args) const
    Calls the delegate if valid.
    For TReturn == void.
    Returns 'true' if the delegate was called, otherwise 'false'.
```

**call_or**
```
template <typename TAlternative>
TReturn call_or(TAlternative alternative, TParams... args) const
    Calls the delegate if valid.
    Calls the 'alternative' if not.
    Run time 'alternative'.
    'alternative' may be a global or static function or a delegate.
```

**call_or**
```
template <TReturn(*Method)(TParams...)>
TReturn call_or(TParams... args) const
    Calls the delegate if valid.
    Calls the 'alternative' if not.
    Compile time 'alternative'.
    'alternative' may be a global or static function.
```

jwellbelove, 2021-09-07:
20.17.0
Added `call_if` and `call_or`

closed 2021-09-07 by jwellbelove