ETLCPP/etl #422: Constexpr Map/Multimap, etc

jovere, 2021-08-31:
Is there a reason Map and its related containers are not `constexpr`? I would like to create a constant lookup table, but it appears that it isn't possible. Is there a different way that could be done?

Thanks,
Jeremy

jwellbelove, 2021-08-31:
Originally the containers were written for C++03, so 'constexpr' was not a thing at the time.
Making all of the ETL containers 'constexpr' is a case of "gradual work in progress".

jwellbelove, 2021-09-01:
What type are your keys to the lookup table?

jovere, 2021-09-01:
Right now I was planning to use just a regular int. Having it as a work in progress is entirely understandable. I wasn't sure if it was a limitation of the data type or some other issue. The stl doesn't have its definition of map as constexpr, either.

jwellbelove, 2021-09-01:
If you were to create a `constexpr etl::array`, of key/value pairs, with the keys in order, then you can do an `etl::lower_bound` search on it.

closed 2021-09-04 by jwellbelove