ETLCPP/etl #423: Trimming nulls from a string_view

jovere, 2021-09-10:
I have a character array that can be a maximum length, but *might* (not guaranteed to) have a null delimiter at the end of the useful data. I have tried multiple ways to turn this into a trimmed string_view without success.

Here is an example of what I have tried. I'd expect to get a `string_view` of `moreData` of length 3 while still making sure I don't go past the end of the array, but it's proving to not be possible.

Is there anything in here that can do what I need?

Thanks,
Jeremy

```c++
#include "etl/string_view.h"
#include "etl/string_utilities.h"
#include <iostream>
char data[8] = { 'h','e','l','l','o','o','o','o' };
char moreData [15] = {'w', 'o', 'r'};

int main()
{
    // Displays 7 (correctly)
    std::cout << "Length with make_string_view " << etl::make_string_view(data).size() << "\n";
    // Displays 11, (BUFFER OVERFLOW!) even though I'm asking for the first string only.
    std::cout << "Length with constructor " << etl::string_view(data).size() << "\n";
    // Displays 14, but has a bunch of nulls at the end (this would seem the best way to go, but doesn't check for null)
    std::cout << "make_string_view " << etl::make_string_view(moreData).size() << "\n";
    // Displays 0, empty string
    std::cout << "trim_view make_string_view " << etl::trim_view(etl::make_string_view(moreData), "\0").size() << "\n";
    // Displays 0, empty string
    std::cout << "trim_view_right make_string_view " << etl::trim_view_right(etl::make_string_view(moreData), "\0").size() << "\n";
    // Displays 14, with the nulls
    std::cout << "trim_from_view_right string_view " << etl::trim_from_view_right(etl::make_string_view(moreData), "\0").size() << "\n";
    // Displays 14, with the nulls
    std::cout << "trim_from_view_left string_view " << etl::trim_from_view_left(etl::make_string_view(moreData), "\0").size() << "\n";
    // Displays 0, empty string
    std::cout << "trim_view_left string_view " << etl::trim_view_left(etl::make_string_view(moreData), "\0").size() << "\n";

    return 0;
}
```

jwellbelove, 2021-09-10:
`etl::string_view(data).size()`
This fails because the constructor is expecting the string to be null terminated. It does find one eventually, but that is only because there happens to be one after characters in the second array.

`etl::string_view(moreData).size() == 3`
This _will_ give the correct result for `moreData`.

The other trim functions cannot work with the delimiter string set to "\0" because "\0" is an empty string. (\0 is a null terminator)

etl::make_string_view from an array make the _whole_ of the array the view.
It may be that a `trim_nulls` could be a useful addition to the string utilities.

jovere, 2021-09-10:
I see that `etl::make_string` appears to do what I want, but it requires making a copy of the buffer, which is undesirable.

jwellbelove, 2021-09-10:
`etl::string_view moreDataView(moreData);` does what you are looking for in the second array's case.

jovere, 2021-09-10:
Just using the plain string_view constructor won't work because if the array is completely filled, it will not find a null terminator. In my application, I can't trust that there will be a null terminator at the end, but if there is one, I don't want to have all the data after it. `etl::find` will probably work for now.

jovere, 2021-09-10:
Also, while looking at the initializer for `etl::make_string`, which takes an array, I see that the call to `strlen` and therefore `length` in `char_traits` does not have an upper count limit. If there was no null terminator, it could count for a very long time, and potentially give a memory access violation.

https://github.com/ETLCPP/etl/blob/5093189a7b1f15bbf0503a4119e0f9ee550491ef/include/etl/char_traits.h#L117-L130

jwellbelove, 2021-09-10:
`etl::char_traits` is designed to be a clone of `std::char_traits` and as such, exhibits the same behaviour.
[https://en.cppreference.com/w/cpp/string/char_traits/length](https://en.cppreference.com/w/cpp/string/char_traits/length)

jovere, 2021-09-10:
If that is the case, wouldn't it be a bad idea to use `strlen` to determine the length of an array, as in `etl::make_string`, since the array being passed in (not a guaranteed c-style string)?

jwellbelove, 2021-09-10:
`etl::make_string` assumes that what is passed to it is a valid (null terminated) C string. This is a fairly standard convention for C string to C++ string conversion in the STL.
If you want to do something different to that then it's should be fairly trivial to put together a function to do what you want.

jovere, 2021-09-10:
Thanks,
I've found a solution that will work:
```
auto end = etl::find(moreData, moreData + sizeof(moreData), '\0');
auto view = etl::string_view(moreData, end);
```

jwellbelove, 2021-09-10:
Good to hear 👍, though you have highlighted the fact that `etl::make_string` and `etl::make_string_view` have slightly different behaviours.
I may have to get some feedback from the ETL community to see whether fixing this (with some extra features to stop buffer overrun) would break existing code.

jovere, 2021-09-10:
Thank you, I appreciate it.

closed 2021-09-12 by jwellbelove