ETLCPP/etl #425: string_view constructors marked as ETL_CONSTEXPR17?

jovere, 2021-09-16:
I'm attempting to use the string_view (and therefore basic_string_view) in a constexpr context as below:

`etl::string_view constexpr myString = "A string";`

In order to do that, I need a constexpr constructor.  I see that `string_view.h` has all the etl::string_view constructors as `ETL_CONSTEXPR17`, but I am required to use C++14. From what I understand of the standard, C++14 should do this just fine. I switched the specifiers to `ETL_CONSTEXPR14` and everything compiles with Clang in C++14 mode and IAR.  Is there something I'm missing with this?

https://github.com/ETLCPP/etl/blob/5093189a7b1f15bbf0503a4119e0f9ee550491ef/include/etl/string_view.h#L120-L169

jwellbelove, 2021-09-16:
Looks like they should be ETL_CONSTEXPR14. I'll do a change.

MickroC, 2021-09-28:
I had the same issue but I'm stuck to C++11, so I replaced `char_traits::length()` by this:

`
ETL_CONSTEXPR static size_t length(const char_type* str, size_t len = 0)
{
  return str == 0 ? 0 : str[len] == '\0' ? len : length(str, ++len);
}
`

would it be possible to integrate it this (or in a similar) way in the library?

jwellbelove, 2022-02-16:
20.24.0

closed 2022-02-16 by jwellbelove