ETLCPP/etl #426: Compilation conflict with Cpputest mock support

DanieleNardi, 2021-09-20:
Hello,
when I try to compile a source file for unit testing that includes both an etl header, like etl/string.h, and the cpputest framework header CppUTestExt/MockSupport.h, I get more than 400 lines of compilation error report from g++. The report is attached to the message.
Compiler: g++
```
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.3.0-17ubuntu1~20.04' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-HskZEa/gcc-9-9.3.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04) 
```
[conflict.txt](https://github.com/ETLCPP/etl/files/7196059/conflict.txt)

What's the cause of this conflict?

jwellbelove, 2021-09-20:
A quick scan of the error messages seems to indicate that CppUTestEx is doing its own #define of 'new'. That would completely mess up the ETL's placement new code.

jwellbelove, 2021-09-20:
Can you try defining `ETL_USING_STD_NEW`

`platform.h` contains these lines to decide the value for that macro.

```
#if !defined(ETL_USING_STD_NEW)
  #if defined(__has_include)
    #define ETL_USING_STD_NEW __has_include(<new>)
  #elif ETL_NOT_USING_STL || (defined(ARDUINO) && defined(__AVR__))
    #define ETL_USING_STD_NEW 0
  #else
    #define ETL_USING_STD_NEW 1
  #endif
#endif
```

closed 2022-01-16 by jwellbelove

Francois-Belanger, 2022-01-28:
Hi, when trying to compile with CPPUtest under OsX I get the following after defining "ETL_USING_STD_NEW" in my CMake.

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++

Compilation stopped after 20 errors like that.

`ETLCPP/include/etl/memory.h:1629:12: error: unknown type name 'p'
    ::new (p) T;
           ^`
Thanks

jwellbelove, 2022-01-28:
I found this on the cpputest Github page.

https://github.com/cpputest/cpputest/issues/742

joreika, 2022-12-22:
Hello, 

I stumbled upon the same problem so I will share how I solved it in hope that it would be helpful. I use ETLCPP in an embedded environment and additionally CppUTest for unit tests in a linux environment. The issue is, as stated also by @jwellbelove , that CppUTest has its own overload of the "new" operator. The reason is that this framework wants to detect memory leaks and has in its file "MemortLeakDetectorNewMacros.h" the following line:

`#define new2 new(__FILE__, __LINE__)`

If one comments it out, then the error should disappear in case 2 things are done:
- using the proper etl_profile.h configuration
- the CppUTest framework is recompiled from source and is used for the tests.

The disadvantage is that one cannot use CppUTest for memory leak detection, but since ETL is used, I do not think this is an issue at all. Here is also my config just in case:

```
#if defined(__EMBEDDED_COMPILER__)

#define ETL_TARGET_DEVICE_GENERIC
#define ETL_TARGET_OS_NONE
#define ETL_CPP11_SUPPORTED 1
#define ETL_CPP14_SUPPORTED 0
#define ETL_CPP17_SUPPORTED 0
#define ETL_NO_NULLPTR_SUPPORT    1
#define ETL_NO_LARGE_CHAR_SUPPORT 1
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED 0
#define ETL_FORCE_NO_ADVANCED_CPP

#else       //for the unit tests

#define ETL_THROW_EXCEPTIONS
#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP
#define ETL_ISTRING_REPAIR_ENABLE
#define ETL_IVECTOR_REPAIR_ENABLE
#define ETL_IDEQUE_REPAIR_ENABLE
#define ETL_TARGET_DEVICE_X86
#define ETL_TARGET_OS_LINUX
#define ETL_COMPILER_GCC
#define ETL_USING_STD_NEW 1

#endif /* __EMBEDDED_COMPILER__ */
```

cheers and thanks for ETL

jwellbelove, 2022-12-22:
Thanks for the update.
I may add more information to the documentation.

joreika, 2022-12-22:
this is great, I did not even thought about looking for that. This would be a better solution for sure, I have to check if it works without issues. Thank you

jwellbelove, 2022-12-22:
Let me know if it works.

joreika, 2022-12-23:
I tried with both `CPPUTEST_USE_MEM_LEAK_DETECTION = N` and `CPPUTEST_MEM_LEAK_DETECTION_DISABLED` defined, but it did not work. It should work with these flags based on their names, but on the other hand none of them disables `#define new new(__FILE__, __LINE__)` when defined so maybe this is the reason

jwellbelove, 2022-12-23:
OK, thanks for checking.

jwellbelove, 2022-12-23:
It looks like you need to adjust the CMake file to disable the macros.

From https://cpputest.github.io/manual.html
Under 'Compiler options'

```
Then for the memory leak detection, you’ll need to add:

    CXXFLAGS += -include $(CPPUTEST_HOME)/include/CppUTest/MemoryLeakDetectorNewMacros.h
    CFLAGS += -include $(CPPUTEST_HOME)/include/CppUTest/MemoryLeakDetectorMallocMacros.h
```

joreika, 2022-12-23:
yes, this finally provides a proper way to disable the memory leak detection, much better than commenting out the line mentioned above. thank you