ETLCPP/etl #431: CMSIS-RTOS2 support

MickroC, 2021-09-28:
Hello,
I'd like to use the etl together with  [CMSIS-RTOS2](https://www.keil.com/pack/doc/CMSIS/RTOS2/html/index.html). For this I had to make some local modifications, which I would like to share here. I would appreciate it if the API were supported by default so that I don't have to maintain my local changes.

in mutex.h:
```
#if ETL_CPP11_SUPPORTED == 1 && ETL_USING_STL
  #include "mutex/mutex_std.h"
  #define ETL_HAS_MUTEX 1
// local modification start
#elif defined(ETL_TARGET_OS_CMSIS_OS2)
  #include "mutex/mutex_cmsis_os2.h"
  #define ETL_HAS_MUTEX 1
// local modification end
#elif defined(ETL_TARGET_OS_FREERTOS)
  #include "mutex/mutex_freertos.h"
  #define ETL_HAS_MUTEX 1
#elif defined(ETL_COMPILER_ARM5) || defined(ETL_COMPILER_ARM6) || defined(ETL_COMPILER_ARM7) || defined(ETL_COMPILER_ARM8)
  #include "mutex/mutex_arm.h"
  #define ETL_HAS_MUTEX 1
#elif defined(ETL_COMPILER_GCC)
  #include "mutex/mutex_gcc_sync.h"
  #define ETL_HAS_MUTEX 1
#elif defined(ETL_COMPILER_GCC)
  #include "mutex/mutex_clang_sync.h"
  #define ETL_HAS_MUTEX 1
#else
  #define ETL_HAS_MUTEX 0
#endif
```

and the actual mutex implementation in mutex/mutex_cmsis_os2.h looks like this:
```
#include <cmsis_os2.h>

namespace etl 
{
  class mutex
  {
  public:
    mutex()
    {
      osMutexAttr_t attr = {"ETL", osMutexRecursive | osMutexPrioInherit | osMutexRobust, 0, 0};
      id = osMutexNew(&attr);
    }

    void lock()
    {
      osMutexAcquire(id, osWaitForever);
    }

    bool try_lock()
    {
      return osMutexAcquire(id, 0) == osOK;
    }

    void unlock()
    {
      osMutexRelease(id);
    }

  private:
    osMutexId_t id{0};
  };
}

```

If this is a possible extension, feel free to use the code as you wish.

jwellbelove, 2021-09-28:
Thanks, I'll add it to the next release.

closed 2021-09-28 by jwellbelove

CurtisAMU, 2023-08-25:
I've modified a copy of this to work with CMSIS-RTOS1 (on RTX) and noticed that there is no destructor. I know that ETL is designed not to use dynamic memory allocation. For my own project I replaced `new`/`delete `with implementations that work on memory pools of fixed-sized blocks that prevent fragmentation. Thus, I may destroy a mutex and would like to call `osMutexDelete(id)` in that moment. Is there a technical reason why a (non-virtual) destructor should be avoided here?
[mutex_cmsis_os1_rtx.zip](https://github.com/ETLCPP/etl/files/12441088/mutex_cmsis_os1_rtx.zip)

jwellbelove, 2023-08-26:
There's no reason why there isn't a destructor for this class. There obviously should be one!
I'll add it to the ETL's version.

Do you know if there is a macro that identifies CMSIS-RTOS1 from CMSIS-RTOS2?

jwellbelove, 2023-08-26:
Is it OK if I use your CMSIS_RTOS1 version in the ETL?

jwellbelove, 2023-08-26:
What is the purpose of `os_mutex_def_etl` & `os_mutex_cb_etl`?
They don't appear to be used anywhere?

```
mutex()
  : os_mutex_cb_etl()
  , id(0)
{
  const osMutexDef_t os_mutex_def_etl = { os_mutex_cb_etl };
  id = osMutexCreate(osMutex(etl));
}
```

The RTOSv1 documentation shows this as a mutex creation example.

```
#include "cmsis_os.h"
  
osMutexDef (MutexIsr); // Mutex name definition
  
void CreateMutex (void)
{
  osMutexId mutex_id;
 
  mutex_id = osMutexCreate(osMutex (MutexIsr));
  if (mutex_id != NULL)  
  {
    // Mutex object created
  }   
}
```

MickroC, 2023-08-26:
`osMutex()` is a macro (see https://www.keil.com/pack/doc/CMSIS/RTOS/html/group__CMSIS__RTOS__MutexMgmt.html#ga1122a86faa64b4a0880c76cf68d0c934)
`osMutex(etl)` expands to `os_mutex_def_etl`

MickroC, 2023-08-26:
One more note: CMSIS-RTOS(1) is just the API! Here RTX ist the actual OS providing this interface and before RTX5 it was just a wrapper around the native RTX OS calls. The API could also be implemented by other OSs (e.g. FreeRTOS). In this case you can not guarantee that the macro expands in the same way!
But: Since the API is so much based on macros for the definition and referencing of resources, I am not sure how to implement the etl-object-oriented approach differently.
Perhaps you should not add support for _CMSIS-RTOS1_ but _RTX-CMSIS-RTOS1_?

CurtisAMU, 2023-08-26:
That's the exact reason I called my file mutex_cmsis_os1_rtx.h. Keil rushed the CMSIS-RTOS version and just wrapped their RTX5 without proper separation.
@jwellbelove Please use whatever you like from my take at CMSIS-RTOS1.

jwellbelove, 2023-08-26:
Are you sure that explicitly declaring `uint32_t os_mutex_cb_etl[4];` is the right way to go?
The documentation appears to say that you should be using `osMutexDef` to declare the control block as this will then match what `osMutex` expands to.

_The parameter to osMutex shall be consistent but the macro body is implementation specific in every CMSIS-RTOS._
_The parameter to osMutexDef shall be consistent but the macro body is implementation specific in every CMSIS-RTOS._

CurtisAMU, 2023-08-26:
osMutexDef is a macro:
```
#if defined (osObjectsExternal)  // object is external
#define osMutexDef(name)  \
extern const osMutexDef_t os_mutex_def_##name
#else                            // define the object
#define osMutexDef(name)  \
uint32_t os_mutex_cb_##name[4] = { 0 }; \
const osMutexDef_t os_mutex_def_##name = { (os_mutex_cb_##name) }
#endif
```

The main problem is `os_mutex_cb_##name[4]`. This is an implementation detail related to RTX5. In CMSIS-RTOS2, the API provides `osRtxMutexCbSize`

MickroC, 2023-08-26:
1.: What happens if `osObjectsExternal` is defined? In that case you would mix individual control block definition with centralized. Can the OS handle this?

2.: As the example shows, the control block is usually defined in global space. Do you have to take special care about the memory alignment if you put it in a class as member variable (`os_mutex_cb_etl[4]`)?

CurtisAMU, 2023-08-26:
To 1.) I think that only helps if you just need to reference OS entities. You still need to define them somewhere the way the underlying OS (RTX) expects. [online doc](https://www.keil.com/pack/doc/CMSIS/RTOS/html/usingOS.html#cmsis_os_h)

To 2.) 4-byte alignment should work fine with this array of uint32_t. That's what I did when I wrapped the CMSIS-RTOS in C++ classes (e.g. Task, Mutex, MessageQueue etc.). I had to place the control blocks into my C++ classes just the way I did for the ETL above, peeking into the macros in cmsis_os.h. I never came across any special alignment in the RTX source (except for the stack and memory pools, which are aligned to 8-bytes).

CurtisAMU, 2023-08-30:
I had missed that question.

Unfortunately, the information API changed too between V1 and V2. In CMSIS-RTOS(1) you can query the macros (#define) osCMSIS for the API version and osCMSIS_RTX for the RTOS kernel.

In CMSIS-RTOS2 there are no macros to evaluate at compile time. You have to call osKernelGetInfo to retrieve the API and kernel versions at run-time. The existence of a typedef named osVersion_t hints at a CMSIS-RTOS2 for that typedef is part of the (current) API specification.