ETLCPP/etl #437: Maybe retrieve conan version from library.properties?

jputcu, 2021-10-01:
I was having issues trying to upload the generated conan package to the server:
`Upload recipe to 'qmax_server' failed: The recipe contains invalid data in the 'scm' attribute (some 'auto' values or missing fields 'type', 'url' or 'revision')`
Maybe it is easier to just use the version from the `library.properties` file?
See https://docs.conan.io/en/latest/howtos/capture_version.html#how-to-capture-package-version-from-text-or-build-files.

I've tried with the following file (rename to py): [conanfile.txt](https://github.com/ETLCPP/etl/files/7265677/conanfile.txt)

jputcu, 2021-10-01:
Doing some test, my attempt with the given conanfile seems to be missing some source files, so better hold off

jwellbelove, 2021-10-01:
I've had some issues trying to get conan working for me.
See #144

jputcu, 2021-10-01:
I've just added the `exports_sources = "include/*"` to the class variables which seems to find the sources, but I think the conan server is having ssl issues.

jwellbelove, 2021-10-11:
That would be great if you could do that.

jputcu, 2021-10-11:
I've created a pull request with a working 20.18.1, now waiting for my username acceptance.