ETLCPP/etl #440: Add std::tuple?

rao000, 2021-10-03:
Would it be possible to add support for std::tuple? It seems like an embedded friendly container since it doesn't allocate.

rao000, 2021-10-03:
Nevermind, just realized variadic templates are a c++11 feature.

closed 2021-10-03 by rao000

jwellbelove, 2021-10-03:
I assume you're using C++03?
There are some parts of the ETL that have a C++03 and C++11 implementations, where the C++03 version has a finite set of templates for 1, 2, 3... etc parameters and the C++11 using variadic parameters.

rao000, 2021-10-03:
I'm actually using c++20 but I know that ETL tries to retain backwards compatibility with c++03 and AFAIK variadic templates are a core language feature so the compiler has to support c++11. I suppose you could do a tuple with overloads for each set of parameters, but that sounds pretty painfull

jwellbelove, 2021-10-03:
Actually, I have added some features to the ETL recently that are not available to C++03 users, as backward compatibility is becoming increasingly more difficult. I think a lot of users have moved over to at least C++11 in the years since I created the ETL.
The hardest features to add are those that require knowledge specific to the compiler.
I've decided to focus more on features that are useful to embedded programmers, rather than trying to create a complete clone of the STL.

rao000, 2021-10-03:
Good to know! There are papers in flight for c++23 that aim to add a lot of things (like tuple) to the list of headers supported in a freestanding environment, hopefully that will remove a lot of the nessesity for cloning and leave you to focus on the things whose implementations need to differ. In the mean time if the ETL implements tuple I'm sure a lot of embedded programmers could use it, although I'm also sure you get way more feature requests than sollicitations for help.

jwellbelove, 2021-10-03:
The original intention for the ETL was to complement the STL, not replace it, but so many companies had a blanket 'No STL' rule, that I had to implement some of the STL for internal use. I then made the reverse engineered versions available, but I then I got a lot of requests for C++11/14/20 STL features, a lot of which are perfectly fine for embedded use in the STL form.

In the last couple of years the ETL has started to resemble a full time job. I added Github sponsorship to the project, but it only brings in a very small income at the moment. I'm not working on a contract at the moment, but if the money starts to run low I may have to work in a paying job for a while. Unfortunately that means that the ETL won't get a lot of attention for at least six months.
It would be nice if I could get some of the large companies using the ETL to dip into their large pockets a little to allow me to keep going full time on the ETL.

rao000, 2021-10-03:
Yeah that's a common theme among FOSS projects unfortunately. I'm curious about how to use the STL in conjunction with ETL. I had thought if you use -ffreestanding you can only safely use libs required by the standard to be in the freestanding implementation. I know an implementation like libstdc++ could chose to allow more but I've found it increadably difficult to find information on exactly what headers -ffreestanding allow me to use. For a concrete example, I'd like to use std::chrono::duration and std::variant (which is fully constexpr, including the destructor) but I'd also like to use etl::vector for its' fixed size

jwellbelove, 2021-10-03:
The ETL and STL can cooperate quite well in that the ETL uses the same naming and functionality conventions that he STL does.
For example, the ETL's iterators work perfectly well with STL algorithms. A lot of the containers use the same name for member functions as their STL counterparts, so a class that worked with `std::vector` will generally work equally well with `etl::vector` or `etl::ivector`.

`constexpr` functionality for the ETL is an ongoing task. Every now and then I spend some time trying to get another part of the ETL `constexpr` compliant.

rao000, 2021-10-03:
Understood, I'll give it a try. As for constexpr, I know the pain. I wrote a bitset class that used std::bit_cast in order to be fully constexpr so I could do things like generate masks for config registers without a bunch of macros. Probably the most time I've spent on a single lib

jwellbelove, 2021-10-03:
Often the difficulty I've had is determining which level of C++ supports the constexpr code I'm writing.
The ETL has macros for ETL_CONSTEXPR, ETL_CONSTEXPR14 and ETL_CONSTEXPR17.  It can take a lot of testing to ensure that the correct one has been applied, especially if I'm reverse engineering a C++17/20 feature where _some_ parts are constexpr valid for C++11 or C++14.