ETLCPP/etl #456: etl::begin()/etl::end() broken?

jovere, 2021-11-02:
I recently tried to update my copy of the ETL to the latest and found that `etl::begin()` and `etl::end()` no longer exist. It appears that in commit f3bf0dc7fe8e1bf04a00d98e78cdf1a78f4d951d they were moved to `iterator.h` and disabled if the standard library is being used. The commit comment states that it's `etl::begin()` and `etl::end()` are being removed when `ETL_NO_STL` is defined, yet it appears to be doing the opposite. Is this intentional? Are the etl specific versions not to be used?

Thanks,
Jeremy

jwellbelove, 2021-11-02:
The commit message is incorrect.
I had to disable the ETL's version when using the STL as there are libraries that use `using` directives with `std::begin` and `std::end` which cause ambiguous definitions between the ETL and STL versions. I had to do the same with `std::swap` vs `etl::swap` a long time ago when I started cloning parts of the STL.

jwellbelove, 2021-11-02:
See issue #448

jovere, 2021-11-02:
Ok, I'll start using `std::end` and `std::begin`.  Thanks.

closed 2021-11-02 by jovere