ETLCPP/etl #458: Feature request: adding make_map, make_array, make_vector, ...

timrid, 2021-11-04:
Hi,

I have the following problem: I would like to create a map with a variable number of entries, where the size of the map is already fixed at initialization. For example like this:
```c++
etl::map<uint8_t, uint32_t, 3> map{ { 0, 1000 }, { 5, 2000 }, { 10, 3000 } };
```

What still bothers me here is that you always have to specify the maximum size of the map as well, and this is not determined automatically. Since C++17 there is a template argument deduction for this, but before C++17 this is unfortunately not possible.

For the `std::array` class there is basically the same problem and [here](https://stackoverflow.com/a/26351623) is a solution by creating a function called `make_array`.

Is this also transferable to an etl::map? So that one could write:
```c++
auto map2 = etl::make_map<uint8_t, uint32_t>( { 0, 1000 }, { 5, 2000 }, { 10, 3000 } );
```

If so, this can certainly be adapted to the other container classes like `etl::make_map`, `etl::make_array`, `etl::make_vector`, ...

jwellbelove, 2021-11-04:
I'll certainly take a look when I get back from Spain in a few days time.

jwellbelove, 2021-11-09:
The closest I've been able to get so far to the syntax you suggested is this...

```
using Pair = std::pair<uint8_t, uint32_t>;

auto map2 = etl::make_map<uint8_t, uint32_t>( Pair{ 0, 1000 }, Pair{ 5, 2000 }, Pair{ 10, 3000 } );
```

jwellbelove, 2021-11-09:
For template deduction...

```
using Pair = std::pair<uint8_t, uint32_t>;

etl::map map3 { Pair{ 0, 1000 }, Pair{ 5, 2000 }, Pair{ 10, 3000 } };
```

jwellbelove, 2021-11-22:
Implemented 20.20.0

closed 2021-11-22 by jwellbelove