ETLCPP/etl #459: etl is in conan!

jputcu, 2021-11-04:
*etl* has just been added to *conan-center*, so you can now do `conan install etl/20.19.0@` (https://conan.io/center/etl).
Perhaps a note could be added to the README?

jwellbelove, 2021-11-04:
Excellent news.
I'll try to get that done when I get back from Spain at the weekend.

jwellbelove, 2021-12-05:
I've tried to install, but I get this error.

```
PS D:\Users\John\Programming\GitHub\etl> conan install etl/20.21.0@
Configuration:
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.runtime=MD
compiler.version=16
os=Windows
os_build=Windows
[options]
[build_requires]
[env]

etl/20.21.0: Not found in local cache, looking in remotes...
etl/20.21.0: Trying with 'conancenter'...
ERROR: HTTPSConnectionPool(host='center.conan.io', port=443): Max retries exceeded with url: /v1/ping (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1129)')))

Unable to connect to conancenter=https://center.conan.io
1. Make sure the remote is reachable or,
2. Disable it by using conan remote disable,
Then try again.
```

jputcu, 2021-12-05:
Same here, hopefully a temporary situation.

jputcu, 2021-12-05:
That version doesn't appear to be at the server, try etl/20.19.0

jwellbelove, 2021-12-05:
How do you get a new version on to the server?

jputcu, 2021-12-05:
As far as I can see, a bot detects new versions. For etl it detected 20.20: https://github.com/conan-io/conan-center-index/issues/3470. Then probably click the "pull request, open one" column. I don't know if this will be automatically or manual, but I don't expect it will be instant.

jputcu, 2021-12-07:
You requested a pull request? The latest version appears to be in Conan, much faster then the first one!

jwellbelove, 2021-12-07:
Yes, I did a PR. It was approved in about 20 minutes or so.

closed 2021-12-23 by jwellbelove