ETLCPP/etl #469: Errors after including etl/variant.h and etl/function.h

sectasy0, 2021-11-20:
I need these two etl libraries in my project, I get a few bugs after compiling. The microcontroller I use is esp8266

`Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcu.html
PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 0.9 (ESP-12 Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP Async WebServer> 1.2.3+sha.1d46269
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Embedded Template Library ETL> 20.11.5
|-- <ESPAsyncTCP> 1.2.2
|   |-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio\build\nodemcu\src\main.cpp.o
In file included from src/libsmarth.hpp:1:0,
                 from src\main.cpp:2:
.pio\libdeps\nodemcu\Embedded Template Library ETL\src/Embedded_Template_Library.h:76:0: warning: "ARDUINO_BOARD" redefined [enabled by default]
        #define ARDUINO_BOARD "Unknown"
 ^
<command-line>:0:0: note: this is the location of the previous definition
In file included from .pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/variant.h:47:0,
                 from src/libsmarth.hpp:2,
                 from src\main.cpp:2:
.pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/placement_new.h: In function 'void* operator new(size_t, void*)':
.pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/placement_new.h:41:14: error: redefinition of 'void* operator new(size_t, void*)'
 inline void* operator new(size_t, void* p) { return p; }
              ^
In file included from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_construct.h:59:0,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_tempbuf.h:60,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:62,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62,
                 from C:\Users\LSD\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238,
                 from src\main.cpp:1:
c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\new:109:14: error: 'void* operator new(std::size_t, void*)' previously defined here
 inline void* operator new(std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
              ^
In file included from .pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/variant.h:47:0,
                 from src/libsmarth.hpp:2,
                 from src\main.cpp:2:
.pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/placement_new.h: In function 'void* operator new [](size_t, void*)':
.pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/placement_new.h:42:14: error: redefinition of 'void* operator new [](size_t, void*)'
 inline void* operator new[](size_t, void* p) { return p; }
              ^
In file included from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_construct.h:59:0,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_tempbuf.h:60,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:62,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62,
                 from C:\Users\LSD\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238,
                 from src\main.cpp:1:
c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\new:111:14: error: 'void* operator new [](std::size_t, void*)' previously defined here
 inline void* operator new[](std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT
              ^
In file included from .pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/variant.h:47:0,
                 from src/libsmarth.hpp:2,
                 from src\main.cpp:2:
.pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/placement_new.h: In function 'void operator delete(void*, void*)':
.pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/placement_new.h:44:13: error: redefinition of 'void operator delete(void*, void*)'
 inline void operator delete(void*, void*) ETL_NOEXCEPT {}
             ^
In file included from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_construct.h:59:0,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_tempbuf.h:60,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:62,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62,
                 from C:\Users\LSD\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238,
                 from src\main.cpp:1:
c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\new:115:13: error: 'void operator delete(void*, void*)' previously defined here
 inline void operator delete  (void*, void*) _GLIBCXX_USE_NOEXCEPT { }
             ^
In file included from .pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/variant.h:47:0,
                 from src/libsmarth.hpp:2,
                 from src\main.cpp:2:
.pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/placement_new.h: In function 'void operator delete [](void*, void*)':
.pio\libdeps\nodemcu\Embedded Template Library ETL\src/etl/placement_new.h:45:13: error: redefinition of 'void operator delete [](void*, void*)'
 inline void operator delete[](void*, void*) ETL_NOEXCEPT {}
             ^
In file included from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_construct.h:59:0,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_tempbuf.h:60,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:62,
                 from c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62,
                 from C:\Users\LSD\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:238,
                 from src\main.cpp:1:
c:\users\lsd\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2\new:116:13: error: 'void operator delete [](void*, void*)' previously defined here
 inline void operator delete[](void*, void*) _GLIBCXX_USE_NOEXCEPT { }
             ^
*** [.pio\build\nodemcu\src\main.cpp.o] Error 1`

etl_profile.h

`#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__

#include <limits.h>
//#define ETL_THROW_EXCEPTIONS
#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP
#define ETL_LOG_ERRORS 1
//#define ETL_8BIT_SUPPORT 1
//#define CHAR_BIT 8
//#define ETL_NO_NULLPTR_SUPPORT 0

#define ETL_TARGET_DEVICE_GENERIC
#define ETL_TARGET_OS_NONE
#define ETL_COMPILER_GCC
#define ETL_CPP11_SUPPORTED (__cplusplus >= 201103L)
#define ETL_CPP14_SUPPORTED (__cplusplus >= 201402L)
#define ETL_NO_NULLPTR_SUPPORT !ETL_CPP11_SUPPORTED
#define ETL_NO_LARGE_CHAR_SUPPORT !ETL_CPP11_SUPPORTED
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED ETL_CPP14_SUPPORTED
#define ETL_ATOMIC_SUPPORTED ETL_CPP11_SUPPORTED

#endif`

jwellbelove, 2021-11-21:
Are you able to check which lines are enabled here, in `platform.h`?

```
// Figure out if we can use the standard library <new> header, if haven't already done so in etl_profile.h
#if !defined(ETL_USING_STD_NEW)
  #if defined(__has_include)
    #define ETL_USING_STD_NEW __has_include(<new>)
  #elif ETL_NOT_USING_STL || (defined(ARDUINO) && defined(__AVR__))
    #define ETL_USING_STD_NEW 0
  #else
    #define ETL_USING_STD_NEW 1
  #endif
#endif
```

Could you try defining `ETL_USING_STD_NEW`

sectasy0, 2021-11-21:
Okay, I figured out the problem, arduino.h was included in main.cpp

closed 2021-11-21 by jwellbelove