ETLCPP/etl #472: Feature request: LRU cache

positron96, 2021-11-23:
All in the title. Considering the great assortment of utility containers and algorithms in ETL, it would be nice to have some cache of fixed size that automatically delets old entries.

Or, better yet, extend the framework to caches with different eviction policies.

jwellbelove, 2021-11-23:
I actually started looking at cache containers several years ago, but never got further than an interface (etl::icache) and a simple LRU cache example at the time. 
Maybe it's time to look at it again. 👍

positron96, 2021-11-23:
Thanks for prompt response!

Upon looking at `icache.h` I've realized that we are talking about different caches. What I mean is a generic data container that is basically a `map` of fixed size that deletes oldest (oldest read / first inserted, whatever) values when inserting new value over capacity (API of such container is basically a get/put/contains set of methods). `icache.h` seems to be a caching IO system that caches data from some backend via write_store/read_store.

Examples of a generic memory cache can be seen in 
* Android [LRUCache](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/util/LruCache.java) / Java [LinkedHashMap](https://docs.oracle.com/javase/8/docs/api/java/util/LinkedHashMap.html#removeEldestEntry-java.util.Map.Entry-)
* python's [functools](https://github.com/python/cpython/blob/main/Lib/functools.py#L479) or [cachetools](https://github.com/tkem/cachetools/blob/master/src/cachetools/__init__.py#L201)

Most implementations contain a map that stores insertion order (e.g. via a double-linked list)

positron96, 2021-11-23:
When there is a general purpose memory cache, it can be (almost trivially) used to build an IO cache that wraps backend.

jwellbelove, 2021-11-23:
Good point