ETLCPP/etl #478: push regularly to https://github.com/ETLCPP/etl-arduino ?

jerabaul29, 2021-12-04:
It looks like https://github.com/ETLCPP/etl-arduino is a bit outdated; any reason for not pushing to it / should / could a github action workflow or similar be set up to periodically push to it? :) .

jwellbelove, 2021-12-06:
The Arduino version of the ETL has a slightly different file layout (mandated by the Arduino platform), and doesn't include the unit tests. I already have a python script that copies the relevant code into another directory, so I should probably get into the habit of running it for every release.

jerabaul29, 2021-12-06:
That would be amazing :) .

jerabaul29, 2021-12-06:
Feel free to close :) . Wonder if there is a way to automate the running of the script and pushing to the arduino repository when a release is performed? :).

jwellbelove, 2021-12-06:
It's much easier for me to just remember to go to the Arduino directory and type `python .\create_arduino_library.py`
The only issue that may occur with the Arduino version is that I don't really have the time to do any extensive testing to check for compatibility. I test with VS2019 & clang on Windows and GCC & clang on Linux (with permutations of STL/No STL C++03, using compiler 'built-ins' etc), but the Arduino platform does have the occasional 'quirk'.

closed 2021-12-06 by jwellbelove