ETLCPP/etl #479: etl::string_stream double formatting broken

LuskeyNoah, 2021-12-10:
I just upgraded release `20.2.2` to `20.21.0` and have noticed that writing a double to a string stream using `precision` has changed to a broken behavior.

I've narrowed it down to the following section of our code.

```c++
    WW_LOG_TRACE("value=%f", value);
    etl::string<20> data_string;
    etl::string_stream data_stream(data_string);
    etl::format_spec format;
    format.precision(10);
    data_stream << format << value;
    WW_LOG_TRACE("data_string=%s", data_string.c_str())
```

Produces the following logs

```
18: [2021-12-09 16:49:05.308] [firmware] [trace] [utility/system] __write_double:292 value=5.987000
18: [2021-12-09 16:49:05.308] [firmware] [trace] [utility/system] __write_double:304 data_string=5.139173455
```

The problem is that the value `5.987000` should be formatted to a string `5.9870000000`.

LuskeyNoah, 2021-12-10:
I have narrowed down the introduction of the bug to the following commit

df29374c6ea5cd48173d8779ba1c46ace4a553dc

Using the following unit test

```c++
    TEST(test_format_stream_float)
    {
      const double value = 1.2345;

      String str;
      Stream ss(str);

      Format format;
      format.precision(10);
      ss << format << value;

      CHECK_EQUAL(String(STR("1.2345000000")), str);
    }
```

```
etl/test/test_string_stream.cpp:326:1: error: Failure in test_format_stream_float: Expected 1.2345000000 but was 1.0330660338
```

Will create a PR if I can find the root cause

LuskeyNoah, 2021-12-10:
The root cause is the change to `uint32_t` on the following line: https://github.com/ETLCPP/etl/blob/e1e3e7a194239fff49697dd80163dd0eb42da46d/include/etl/private/to_string_helper.h#L323

When the precision value is 10 or above, this value overflows.

I've added a proposed fix to the PR #480

jwellbelove, 2021-12-23:
Fixed 20.22.0

PR #480 (2021-12-10): Fix for double formatting with precision > 9
- Resolves #479
closed 2021-12-23 by jwellbelove