ETLCPP/etl #481: etl::span<const Data> can not be created from non-const C-array of Data

mnaydenov, 2021-12-13:
```
#include <etl/span.h>
#include <span>

bool f(etl::span<const char, 12> val) //< change to std::span to make it compile
{}

int main()
{
  char c[12];
  
  f(c);
  
  return 0;
}
```

The problem is, the constructor "from a container or other type that supports data() and size() member functions" is incorrectly picked.

jwellbelove, 2021-12-13:
Does it work with `std::span`?

mnaydenov, 2021-12-13:
> Does it work with `std::span`?

Yes.

mnaydenov, 2021-12-14:
`etl::array_view` also has the same issue.

jwellbelove, 2021-12-14:
It looks like I have to disable `ETL_CONSTEXPR span(TContainer& a) ETL_NOEXCEPT` if `TContainer` is an array.
This can only be done for C++11 or above.

mnaydenov, 2021-12-14:
Hello, and thank You for Your efforts. Isn't it possible to SFINAE-enable the overload only if the container has `data()` and `size()` members. I think, a trait for that is possible in beseline C++. I might look into it myself in the following days.

jwellbelove, 2021-12-14:
I can also get it to work by replacing data() & size() with etl/std::begin() and end(), then it should be OK for all versions of C++.

jwellbelove, 2021-12-14:
```
#if ETL_CPP11_SUPPORTED && ETL_USING_STL
  #include <array>
  #include <iterator>

  #define ETL_BEGIN(x) std::begin(x)
  #define ETL_END(x)   std::end(x)
#else
  #include "iterator.h"

  #define ETL_BEGIN(x) etl::begin(x)
  #define ETL_END(x)   etl::end(x)
#endif
```

```
    template <typename TContainer>
    ETL_CONSTEXPR span(TContainer& a) ETL_NOEXCEPT
      : mbegin(etl::addressof(*ETL_BEGIN(a)))
      , mend(etl::addressof(*ETL_END(a)))
    {
    }
```

jwellbelove, 2021-12-23:
Fixed 20.22.0

closed 2021-12-23 by jwellbelove