ETLCPP/etl #482: Two or more etl::span/array_view of different types create ambiguous overloading set

mnaydenov, 2021-12-14:
```
#include <etl/span.h>
#include <span>
#include <etl/array_view.h>

struct C{};

void f(etl::span<char> )
{}

void f(etl::span<C> )
{}

int main()
{
  etl::array<C, 12> c;
  
  f(c);
  
  return 0;
}
=====
error: call to 'f' is ambiguous
  f(c);
  ^
<source>:11:6: note: candidate function
void f(etl::span<char> )
     ^
<source>:14:6: note: candidate function
void f(etl::span<C> )
```
Same issue with `etl::array_view`. Works as expected with `std::span`..

jwellbelove, 2021-12-14:
I've been looking this for a couple of hours and it seems a bit tricky to solve.
It will have an answer, as `std::span` works, but it's going to take a while to work out.
I may be taking a new contract soon (I can't work full time on the ETL and live on savings for much longer!) so I'm not sure when this will get done (my contracts are usually 3 to 6 months or more).

The workaround of course is to explicitly construct the span before passing to the functions.

jwellbelove, 2021-12-23:
Fixed 20.22.0

closed 2021-12-23 by jwellbelove