ETLCPP/etl #483: Flood of unrecognized #pragma "minmax_push.h",  "minmax_pop.h" using ETL_COMPILER_GREEN_HILLS

mnaydenov, 2021-12-14:
The title say it all, `ETL_COMPILER_GREEN_HILLS` does not recognize `push_macro`, `pop_macro`.
This issue prevents us from migrating away from version 18.xx as code audit fails.

jwellbelove, 2021-12-14:
Does the Greenhills compiler support any sort of macro push/pull?

mnaydenov, 2021-12-15:
> Does the Greenhills compiler support any sort of macro push/pull?

Just checked the manual. I think, it is safe to say, there isn't.

jwellbelove, 2021-12-16:
That should be an easy fix as it just requires  `&& !defined(ETL_COMPILER_GREEN_HILLS)` to be added to the `#if` in the headers.

jwellbelove, 2021-12-23:
Fixed 20.22.0

mnaydenov, 2022-01-07:
Hello, back at work now, I tested the latest version and now the build fails because `min` and `max` are `#defined`, breaking `std::numeric_limits`.
If I comment out the new defines it all compiles fine. Do we need these defines? Or more specifically, when do we need them?
As mentioned, I am trying to migrate our codebase from version 18-something, which compiles fine on GHS. 
Thank You.

jwellbelove, 2022-01-07:
It looks like, for GHS, min and max are not macros. The problem normally arises because min & max are usually macros, and as macros ignore namespaces, they screw up std::min/max and etl::min/max. For most every compiler I've tested with so far have required that the macros are temporarily disabled when the STL/ETL versions are used.

jwellbelove, 2022-01-07:
Can you try substituting these files?
[min-max.zip](https://github.com/ETLCPP/etl/files/7828641/min-max.zip)

mnaydenov, 2022-01-07:
Looks good, at least for our project.

jwellbelove, 2022-01-08:
Good, I'll include the change in the next release.

closed 2022-01-08 by jwellbelove