ETLCPP/etl #485: etl::message_packet takes message types that are not in it's list

jovere, 2021-12-15:
I've come across an issue with the `etl::message_packet` and perhaps the `etl::message` framework in general.  In the following code, ([linked at compiler-explorer as well](https://compiler-explorer.com/z/6rxTETb4s)), I can accidentally put a message from my main system messages into my state machine messages and the compiler, nor the running program will have any issue with it.

I believe a couple different factors are allowing this to occur:
1. Rather than taking a reference to an `imessage`, the `message_packet` needs a constructor for each individual type in the packet, without the `static_cast` presently used in the `add_new_message` function. The `add_new_message` function also needs to take the real type as well. This would do the compile time checking to make sure the packet matches.
2. Using a `uint8_t` as the underlying index into the `imessage`s allows one to swap messages between systems (say two state machines) without any idea this is happening. Having an `enum` or `enum class` as the underlying type would force the compiler to check if the messages belong to the same "family". Unfortunately, I think the `imessage` would need to be templated itself in order for this to work, which would potentially break existing code.

```C++
#include "etl/message_bus.h"
#include "etl/message_packet.h"
#include "fmt/format.h"

namespace main_system {
struct main_message : public etl::message<1> {
    explicit main_message(uint32_t data) : data(data) {}
    uint32_t data;
    uint32_t moreData[34];
};
}  // namespace main_system

namespace state_machine {
struct sm_message : public etl::message<1> {
    explicit sm_message(uint8_t data) : data(data) {}
    uint8_t data;
};
using packet_t = etl::message_packet<sm_message>;
}  // namespace state_machine

int main() {
    main_system::main_message mySystemMessage{0xFFFFFFFF};

    state_machine::packet_t myPacket{
        mySystemMessage};  // This should fail, but doesn't

    fmt::print("{}", myPacket.get().get_message_id());  // Invalid data
}
```

jwellbelove, 2021-12-15:
I see the constructors acting like this.
There should still be an `etl::imessage` as well as a `TMessage` constructor (backwards compatibility), but the `etl::imessage` constructor would fail at runtime (`ETL_ASSERT`) and the `TMessage` would fail at compile time due to a missing constructor (modified `etl::enable_if` condition, using `etl::is_one_of`)

jwellbelove, 2021-12-15:
OK, the `etl::imessage` constructor has precedence over the templated constructor, so an invalid message types is not rejected at compile time, so the `enable_if` is not going to work.

jwellbelove, 2021-12-23:
Fixed 20.22.0

closed 2021-12-23 by jwellbelove

jwellbelove, 2021-12-24:
I have kept the `etl::imessage` constructors. An unhandled message type will sent to this constructor will emit an `etl::unhandled_message_exception` error.
Unhandled concrete message types will result in a compile time error.