ETLCPP/etl #486: etl::span of fixed size can be created from an array of a different size

mnaydenov, 2021-12-15:
```
#include <etl/span.h>
#include <span>

void f(etl::span<char, 25> )
{}

int main()
{
  std::array<char, 10> c;
  
  f(c);
  
  return 0;
}
```
The above code should **not** compile. 
The major attraction of using `span` instead of `array_view` is the fact, one can have it with fixed size, enforceable in compile time. This way if a function accepts a serial number of, say 25 digits, it can force the caller to give it only strings of this size. Currently however this enforcement does not work and the above code compiles. Using `std::span` works as expected and it fails to compile.

jwellbelove, 2021-12-23:
Fixed 20.22.0

jwellbelove, 2021-12-23:
Fixed 20.22.0

closed 2021-12-23 by jwellbelove