ETLCPP/etl #491: Bit shifting causes compiler warning with IAR compiler

kreilic, 2022-01-13:
Compiling the ETL with the IAR ARM 8.50.8 compiler causes a compiler warning in a bit shift operation in binary.h file in line 611.

`static ETL_CONSTANT uint32_t value = uint32_t((value4 >> 16U) | (value4 << 16U));`

Here value4 is shiftet by 16 bits to the left, which of course leads to an overflow if value4 is an uint32_t. My prosposal is to cast value4 to uint64_t and shift it afterwards. At the end it is casted back to uint32_t anyway.

jwellbelove, 2022-01-13:
It can't be cast to 64 bit because 64bit integers can be very slow or non-existent on some processor architectures.

Can you try this?
`static ETL_CONSTANT uint32_t value = uint32_t((value4 >> 16U) | ((value4 & 0xFFFFU) << 16U));`

kreilic, 2022-01-14:
Yes you're right, I have not taken this into account.

Your proposal is also working fine. The compiler warning is fixed.

jwellbelove, 2022-01-14:
Great, I'll update the code.

kreilic, 2022-01-14:
That's really great. Thanks a lot for your support.

jwellbelove, 2022-01-23:
20.23.0

closed 2022-01-23 by jwellbelove