ETLCPP/etl #492: Arduino does not support nan/f/l

Mipsters, 2022-01-14:
I'll admit ahead that I probably don't understand the difference between Arduino platforms and compilers enough
I use Arduino mega with `avrisp`, but I don't know if that makes a difference

`etl\limits.h` has this section

```
#if defined(ETL_NO_CPP_NAN_SUPPORT)
  #if defined(NAN)
    #define ETL_NAN  (double)NAN
    #define ETL_NANF (float)NAN
    #define ETL_NANL (long double)NAN
    #define ETL_HAS_NAN true
  #else
    #define ETL_NAN  (double)0.0
    #define ETL_NANF (float)0.0
    #define ETL_NANL (long double)0.0
    #define ETL_HAS_NAN false
  #endif
#else
  #define ETL_NAN  nan("")
  #define ETL_NANF nanf("")
  #define ETL_NANL nanl("")
  #define ETL_HAS_NAN true
#endif
```

in my Arduino env, it uses the last one, because `ETL_NO_CPP_NAN_SUPPORT` is not defined
but Arduino does not support those functions in its `math.h`

the best solution will probably be in `etl\profiles\determine_compiler_language_support.h`
there is this code section
```
// NAN not defined or Rowley CrossWorks
#if !defined(NAN) || defined(__CROSSWORKS_ARM) || defined(ETL_COMPILER_ARM5)
  #define ETL_NO_CPP_NAN_SUPPORT
#endif
```

Arduino does support `NAN`, so it seems to assume it supports all other `nan/f/l`
so I don't know if its best to add `|| defined(ARDUINO)` or `avr`
but I think I laid the problem well enough for you to make a decision :)

Mipsters, 2022-01-22:
This issue should probably be closed after updating the library in Arduino's registry?

jwellbelove, 2022-01-23:
20.23.0

closed 2022-01-23 by jwellbelove