ETLCPP/etl #493: Feature request: discontinuous array_view or span for base class types

PetteriAimonen, 2022-01-16:
I have an array of subclasses, and I would like to offer a view to an array of base classes, without exposing the SubClass implementation:

```
class BaseClass { ... };
class SubClass: public BaseClass { ... };
SubClass myarray[5];

some_array_view<BaseClass> get_array_view()
{
     return some_array_view<BaseClass>(myarray);
}
```

If my understanding is correct, this could be implemented by storing:

- `mitemsize = sizeof(SubClass)`
- Pointer to first element cast as `BaseClass` (to get the possible multi-inheritance offset): `mbegin = static_cast<BaseClass*>(&myarray[0])`
- Number of items in array

and then item retrieval would be something like:

```
T& some_array_view::at(size_t i) {
   return *reinterpret_cast<T*>(reinterpret_cast<char*>(mbegin) + mitemsize * i);
}
```

Open questions:

- Does this make sense?
- Does something like this already exist?
- What should it be called?

jwellbelove, 2022-01-16:
An interesting idea, though most people just go with an array of pointers to the base, which you could pass as a 'span of pointers'.

To do this there would have to be two span classes defined, so as to be able to define the subclass size, but also to negate having to pass this information to the code that wants to iterate through the bases.

The containers use a similar technique. You create a container of a specific size, `etl::vector<int, 10>` but you pass it to code that expects any sized vector by a reference to an `etl::ivector<int>`.

For you proposal you would need something like this...

```
template <typename TBase>
class ibase_class_span
{
  ...
};

template <typename TBase, typename TSubclass>
class base_class_span : public ibase_class_view<TBase>
{
  ...
};

class BaseClass { ... };
class SubClass: public BaseClass { ... };
SubClass myarray[5];

void Function(ibase_class_span<BaseClass>& bcs)
{
  std::for_each(bcs.begin(), bcs.end(), [](BaseClass& b){ b.DoSomething(); });
}

base_class_span<BaseClass, SubClass> bcs(myarray, 5);

Function(bcs);
```

PetteriAimonen, 2022-01-16:
I agree that the common solution to the problem is passing an array/vector of pointers. On PC for such situations I usually have a vector of smart pointers. But on embedded side I prefer static allocation (which ETL is great for!), so I store classes in the array directly. To get the array of pointers I would then need to generate that on request, or to separately maintain it in sync with the main container.

I actually found an existing implementation: https://github.com/mrshurik/poly_span

jwellbelove, 2022-01-23:
An interesting implementation, it looks like he calculates the element size when the span is constructed, rather then when compiled.

jwellbelove, 2022-07-01:
I've downloaded the code for `poly_span`.
As it has also has an MIT licence I have decided to modify its code to merge with the ETL.
Modifications include removal of the `std::initializer_list` constructors (results in pointers to data that may be temporary), and support for `etl::dynamic_range`, as in the standard `etl::span`.

jwellbelove, 2022-07-14:
I didn't use the code from https://github.com/mrshurik/poly_span in the end as it wasn't straight forward to adapt it to the STL and ETL `span` API.
`etl::poly_span` is now in [20.31.0](https://github.com/ETLCPP/etl/releases/tag/20.31.0)

PetteriAimonen, 2022-07-15:
Looks great!