ETLCPP/etl #497: etl::bit_set bugs

jwellbelove, 2022-01-21:
Raised by Thomas Gerstner, Siemens AG

1. etl::bitset<> has wrong size()
This goldbolt example compares the behavior of STL and ETL: https://gcc.godbolt.org/z/rc4fMrEj9
It uses a etl::bitset<5> and negates all bits. The expected size after the negation should still be 5 but it actually is 8.
STL has the correct behavior.

2. etl::bitset<0> segfaults
This goldbolt example compares the behavior of STL and ETL: https://gcc.godbolt.org/z/jovn9WrEb
It's basically the same example as above except that there is a shift operation by 1 to the left.
The cout's are not necessary but you can see that the output is not complete because the program somehow crashes (return code 139).
The exit code 139 is a fatal error signal (128 + 11) and 11 means SIGSEGV.

Mipsters, 2022-01-22:
after the merge this issue can probably be closed?

closed 2022-01-22 by jwellbelove

TG-SAG, 2026-10-13:
I compiled the first example again with a current ETL version (20.49.0) and the `new bitset `did not fix the problem while the `legacy bitset `seems to be correct. Here's a comparison between both bitset implementations: https://gcc.godbolt.org/z/xvc9nhqTh