ETLCPP/etl #498: Greenhills compile error due to etl macro max overwrites std::numeric_limits<uint16_t>::max()

Jogi7819, 2022-01-21:
[ghs_example_bug.zip](https://github.com/ETLCPP/etl/files/7912724/ghs_example_bug.zip)

Affected Version: 20.22.0

The macro declared here https://github.com/ETLCPP/etl/blob/master/include/etl/private/minmax_pop.h#L40 is propagated into client code and overwrite such function calls like

`auto max_uint16_ = std::numeric_limits<uint16_t>::max();`

Compiling the attached example with command

`cxppc  -DETL_ARRAY_VIEW_IS_MUTABLE -DPB_CONVERT_DOUBLE_FLOAT -sys_include_directory external/etl-src/include -language=cxx --stdl --c++14 --brief_diagnostics -o ghs_example_bug.cpp.o -c ./ghs_example_bug.cpp`

leads to the following error message
```
"./ghs_example_bug.cpp", line 9 (col. 59): error #54-D: too few arguments in invocation of macro "max"
"./ghs_example_bug.cpp", line 9 (col. 55): error #40: expected an identifier
"./ghs_example_bug.cpp", line 9 (col. 55): error #29: expected an expression
"./ghs_example_bug.cpp", line 9 (col. 55): error #29: expected an expression
"./ghs_example_bug.cpp", line 9 (col. 55): error #29: expected an expression
"./ghs_example_bug.cpp", line 9 (col. 55): error #29: expected an expression
```

jwellbelove, 2022-01-21:
I am already dealing with this issue.
See. #483

See https://github.com/ETLCPP/etl/issues/483#issuecomment-1007354510 for the interim fix

closed 2022-01-21 by jwellbelove

Jogi7819, 2022-01-21:
Perfect, thank you.