ETLCPP/etl #502: Class template argument deduction not working?

Compizfox, 2022-01-23:
I'm trying to use `etl::overload` to create a functor to visit a `etl::variant`, but I'm having some issues with creating the overload. The following code doesn't compile:

```
auto overload = etl::overload {
	[](uint8_t* x) {return 0;},
	[](uint16_t* x) {return 0;},
	[](double* x) {return 0;}
};
```

The compiler reports the following error:

```
test.cpp:10:25: error: missing template arguments before '{' token
  auto overload = etl::overload {
```

It seems like class template argument deduction is not working?

I'm using gcc-arm-none-eabi 5.4.1 (for a Teensy MCU) with `-std=c++17`.

The version of the ETL I'm using is 20.22.0 (`etlcpp/Embedded Template Library ETL`, from the PlatformIO repository).

jwellbelove, 2022-01-23:
Can you try the alternative syntax?

```
auto overload = etl::make_overload(
	[](uint8_t* x) {return 0;},
	[](uint16_t* x) {return 0;},
	[](double* x) {return 0;});
```

Compizfox, 2022-01-23:
Unfortunately that doesn't compile either:

```
.pio/libdeps/teensylc/Embedded Template Library ETL/src/etl/overload.h: In instantiation of 'constexpr etl::overload<TOverloads ...> etl::make_overload(TOverloads&& ...) [with TOverloads = {<lambda(uint8_t*)>, <lambda(uint16_t*)>, <lambda(double*)>}]':
src/test.cpp:7:28:   required from here
.pio/libdeps/teensylc/Embedded Template Library ETL/src/etl/overload.h:83:76: error: no matching function for call to 'etl::overload<<lambda(uint8_t*)>, <lambda(uint16_t*)>, <lambda(double*)> >::overload(<brace-enclosed initializer list>)'
     return overload<TOverloads...>{ etl::forward<TOverloads>(overloads)... };
                                                                            ^
.pio/libdeps/teensylc/Embedded Template Library ETL/src/etl/overload.h:63:10: note: candidate: constexpr etl::overload<<lambda(uint8_t*)>, <lambda(uint16_t*)>, <lambda(double*)> >::overload(const etl::overload<<lambda(uint8_t*)>, <lambda(uint16_t*)>, <lambda(double*)> >&)
   struct overload : TFirst, overload<TOthers...>
          ^
.pio/libdeps/teensylc/Embedded Template Library ETL/src/etl/overload.h:63:10: note:   candidate expects 1 argument, 3 provided
.pio/libdeps/teensylc/Embedded Template Library ETL/src/etl/overload.h:63:10: note: candidate: constexpr etl::overload<<lambda(uint8_t*)>, <lambda(uint16_t*)>, <lambda(double*)> >::overload(etl::overload<<lambda(uint8_t*)>, <lambda(uint16_t*)>, <lambda(double*)> >&&)
.pio/libdeps/teensylc/Embedded Template Library ETL/src/etl/overload.h:63:10: note:   candidate expects 1 argument, 3 provided

```

jwellbelove, 2022-01-23:
Does your compiler complain about this (which is effectively what etl::overload is doing)

```
      auto b1 = [](uint8_t* x) {return 0; };
      auto b2 = [](uint16_t* x) {return 0; };
      auto b3 = [](double* x) {return 0; };

      struct Overload : decltype(b1), decltype(b2), decltype(b3)
      {
        using decltype(b1)::operator();
        using decltype(b2)::operator();
        using decltype(b3)::operator();
      };
```

jwellbelove, 2022-01-23:
All of the above compile OK here on C++17 using VS2019, GCC 10.3.0 and clang 10.0.0

jwellbelove, 2022-01-23:
```
      auto b1 = [](uint8_t* x) {return 0; };
      auto b2 = [](uint16_t* x) {return 0; };
      auto b3 = [](double* x) {return 0; };

      struct Overload : public decltype(b1), public decltype(b2), public decltype(b3)
      {
        using base1 = std::decay_t<decltype(b1)>;
        using base2 = std::decay_t<decltype(b2)>;
        using base3 = std::decay_t<decltype(b3)>;

        Overload(decltype(b1) b1, decltype(b2) b2, decltype(b3) b3)
          : base1(b1), base2(b2), base3(b3)
        {
        }
      };

      Overload ov(b1, b2, b3);

      ov((uint8_t*)NULL);
```

Compizfox, 2022-01-23:
> Does your compiler complain about this (which is effectively what etl::overload is doing)
>
> ```
>       auto b1 = [](uint8_t* x) {return 0; };
>       auto b2 = [](uint16_t* x) {return 0; };
>       auto [...]

Nope, it doesn't complain about this.

jwellbelove, 2022-02-16:
I don't know if you've already got to the bottom of this, or used another method, but does this work?

```
#include <cstdint>
#include <type_traits>

auto b1 = [](uint8_t* x) {return 0; };
auto b2 = [](uint16_t* x) {return 0; };
auto b3 = [](double* x) {return 0; };

struct Overload : public decltype(b1), public decltype(b2), public decltype(b3)
{
  using base1 = std::decay_t<decltype(b1)>;
  using base2 = std::decay_t<decltype(b2)>;
  using base3 = std::decay_t<decltype(b3)>;

  using base1::operator();
  using base2::operator();
  using base3::operator();

  Overload(decltype(b1) b1, decltype(b2) b2, decltype(b3) b3)
    : base1(b1), base2(b2), base3(b3)
  {
  }
};

int main()
{
  Overload ov(b1, b2, b3);

  uint8_t* p = 0;

  ov(p); // Calls uint8_t* lambda
}
```

jwellbelove, 2022-02-21:
I'm going to close this.
If you still have issues, then please re-open.

closed 2022-02-21 by jwellbelove

Compizfox, 2022-02-21:
Sorry, I haven't really pursued this issue any further in the meantime.

> I don't know if you've already got to the bottom of this, or used another method, but does this work?
>
> ```
> #include <cstdint>
> #include <type_traits>
>
> auto b1 = [](uint8_t* x) {return 0; };
> auto b2 = [...]

This compiles. I guess the issue with the previous snippet was just a function call outside a function body.