ETLCPP/etl #506: ETL_NULLPTR issue with Code Composer Studio ( C++ 03 )

RallyTronics, 2022-02-08:
When compiling with Code Composer Studio for the C2000, the compiler does not seem to cast ETL_NULLPTR to the desired type.  This can be seen by instantiating an etl scheduler.  I get the following errors:
```
include/etl\scheduler.h", line 326: error #145: a value of type "void *" cannot be used to initialize an entity of type "etl::ifunction<void> *
include/etl\scheduler.h", line 327: error #145: a value of type "void *" cannot be used to initialize an entity of type "etl::ifunction<void> *

at this point in the scheduler constructor
` p_idle_callback(ETL_NULLPTR),
  p_watchdog_callback(ETL_NULLPTR),`

```
If I redefine ETL_NULLPTR  to 0 rather than NULL the problem is fixed

jwellbelove, 2022-02-09:
What standard of C++ are you using?

jwellbelove, 2022-02-09:
It looks like Code Composer may be using the C definition of NULL.
Visual Studio's runtime defines NULL as this...
```
#ifndef NULL
    #ifdef __cplusplus
        #define NULL 0
    #else
        #define NULL ((void *)0)
    #endif
#endif
```

RallyTronics, 2022-02-09:
For C2000 micros, C++03 is about the only option.

jwellbelove, 2022-02-10:
I've decided to set `ETL_NULLPTR` to `0` for `ETL_CPP11_NOT_SUPPORTED || ETL_NOT_USING_STL`

closed 2022-02-10 by RallyTronics