ETLCPP/etl #508: Compiling under C++20 gives an error

poke1024, 2022-02-13:
Not sure if C++20 is officially supported right now.

When compiling through clang-1300.0.29.30 with C++20, the following error occurs:

```
.../etl/include/etl/memory.h:979:30: error: use of undeclared identifier 'o'
    return std::construct_at(o, etl::forward<TArgs>(args)...);
```

Looking at the code, there seems to be a missing definition for `o`?

https://github.com/ETLCPP/etl/blob/9ff3cd095ab6d6a2c064aa6d4e5899f7f6f51f0c/include/etl/memory.h#L976-L980

poke1024, 2022-02-13:
For now, I worked around by using `#define ETL_CPP20_SUPPORTED 0`.

jwellbelove, 2022-02-13:
There's a typo. the `o` should be `p`.
This will be fixed in the next release.

closed 2022-02-13 by jwellbelove