ETLCPP/etl #509: Incorrectly including initializer_list on AVR

jputcu, 2022-02-22:
Upgrading to `20.24.0` from `20.21.0` I get compiler errors related to `initializer_list`.
~~~~
include\etl\initializer_list.h(43,11): error: initializer_list: No such file or directory
		  #include <initializer_list>
		           ^~~~~~~~~~~~~~~~~~
~~~~

jwellbelove, 2022-02-22:
Does AVR not support initializer_list with C++11 and STL?

jputcu, 2022-02-22:
AVR doesn't have the C++ headers, nor STL. I'm using the AVR cross compiler from Arduino.

jwellbelove, 2022-02-22:
Ah, Arduino. The nemesis of the ETL!

Try defining `ETL_NO_STL` in your project properties, or `etl_profile.h`, if you have one.

jputcu, 2022-02-22:
I have that define already present:
`-DETL_TARGET_DEVICE_GENERIC -DETL_TARGET_OS_NONE -DETL_NO_STL -DETL_NO_CPP_NAN_SUPPORT -DETL_NO_HUGE_VAL_SUPPORT -DETL_USE_LEGACY_VARIANT`

jwellbelove, 2022-02-22:
Your setup should make this pre-processor directive fail.

`#if ((ETL_USING_STL && ETL_NOT_USING_STLPORT) || defined(ETL_IN_UNIT_TEST)) && !defined(ETL_IN_UNIT_TEST_INITIALIZER_LIST)`

which should stop <initializer_list> being included.

jwellbelove, 2022-02-22:
That should equate to...

`#if ((0 && 0) || 0) && 1` which is `0`

jputcu, 2022-02-22:
There is a mix of just the defines and `defined`, maybe that causes some issues

jputcu, 2022-02-22:
Ok, with your defines it works, but with `20.24.0`, I have `(ETL_USING_STL && ETL_NOT_USING_STLPORT) || !defined(ETL_IN_UNIT_TEST_INITIALIZER_LIST)`

jwellbelove, 2022-02-22:
OK, I understand now! You need `20.24.1`

jputcu, 2022-02-22:
That seems to work

closed 2022-02-22 by jputcu