ETLCPP/etl #512: `etl::intrusive_list` constructor could be `constexpr`

diggit, 2022-03-03:
As title says, I think `etl::intrusive_list` constructor could be `constexpr` ot at least adding `constexpr` specifier does not seem to cause issue in my case. This may also apply to other member functions.
Is there a reason, why they are not marked `constexpr`?

jwellbelove, 2022-03-03:
There are a lot of containers (and other code) that could be `constexpr`. 
There are a few reasons why this is not so (yet).

1. The library was originally coded for C++03, as a lot of embedded work didn't use anything above that at the time the ETL was created.
2. Some projects still don't use anything above C++03 as they are limited to MISRA 2008 rules.
3. There are over 50 classes that deal with containers alone, and most are not `constexpr` yet (for the above reasons).
4. Maintaining the ETL doesn't pay the bills, so I only do this whenever I get spare time.
5. There are enough potential updates for the ETL to be a 'day job'.
6. There is only one of me.

I'm not saying that it will not get done at some point, just that there are no fixed deadlines for completion.

closed 2022-03-04 by jwellbelove

diggit, 2022-03-04:
Oh, I am surprised, that project like ETL does not receive more funding from users.
Creating issues with some todos and marking them with difficulty tags including things like "beginner friendly" could help to bring more contributors.

This seems to be only matter of adding `ETL_CONSTEXPR` or am I missing something (new tests etc.).
I think, I can help with that.

jwellbelove, 2022-03-04:
Some containers may be fairly easy to convert to `constexpr`. They would all need unit tests to confirm that each member function compiles correctly in a `constexpr` context. Depending on what the each member function does or how it's coded, `constexpr` may only be available for C++14, so `ETL_CONSTEXPR14` would be used instead. Some may only be `constexpr` if the underlying copy algorithms are also `constexpr`. For fundamental types, such as integers, the copy algorithm will try to use a fast 'memcpy'. To be `constexpr`, the algorithm must either use the compiler's intrinsics or a simple loop copy. This will depend on the availability of the intrinsics, or the user setting forcing `constexpr` algorithms.
It can get a little complicated!