ETLCPP/etl #513: Broken web style

diggit, 2022-03-04:
There seems to be issue with styling of www.etlcpp.com
Not every page is affected
Tested with Firefox 98 (developer build)  and Chromium 99.0.4844.51 with same results.
[image]
[image]
[image]
[image]

Here is non extensive list of affected pages:
- https://www.etlcpp.com/instance_count.html
- https://www.etlcpp.com/intrusive_forward_list.html
- https://www.etlcpp.com/queue.html
- https://www.etlcpp.com/observer.html
- https://www.etlcpp.com/visitor_tutorial.html
- https://www.etlcpp.com/observer_tutorial.html
- https://www.etlcpp.com/message_tutorial.html
- https://www.etlcpp.com/generators_tutorial.html

Unfortunately I am complete CSS noob, so I am unable to offer fix.

jwellbelove, 2022-03-04:
The pages aren't coded in raw HTML. I use a WYSIWYG editor to create the web site, but it does seem to have output rendering issues at times. I'll see if I can persuade it to behave!

closed 2022-03-23 by jwellbelove

diggit, 2022-06-27:
Stumbled on more:
- https://www.etlcpp.com/function.html
- https://www.etlcpp.com/parameter_type.html