ETLCPP/etl #515: Request to make unaligned types fully constexpr

benedekkupper, 2022-03-11:
As a follow-up of #489 I'd like to ask for these unaligned types to be made fully `constexpr`. Here is one good example of what could be used:
http://stackoverflow.com/a/36937049

jwellbelove, 2022-03-12:
As a `constexpr` function cannot cast a pointer, many functions in unaligned type can only be `constexpr` if an intrinsic for `memcpy` or equivalent is available. I know that recent GCC and Clang compilers support `__builtin_memcpy` (MSVC doesn't), but I'm not sure about others.

benedekkupper, 2022-03-13:
How about using arithmetic operations (so shifting and masking / shifting and bitwise ORing) to convert between bytes and the native type instead of memory copying? That would be inherently `constexpr`. The question is if there is a performance difference between the arithmetic operations and memory operations in runtime, but this could be profiled, if the compilers are smart enough to recognize what the arithmetics are doing.

jwellbelove, 2022-03-13:
That's a good point. I've been too focussed on `memcpy` as the only solution.

jwellbelove, 2022-03-23:
v20.26.0

benedekkupper, 2025-05-16:
So I just noticed that there's a regression with v20.40.0, specifically 20ef1a3 and earlier, what was the motivation to do this?