ETLCPP/etl #518: GCC not properly generating optimal code for circular_buffer

flaviut, 2022-03-26:
Hey, I'm not seeing optimal code for `circular_buffer.pop()` with GCC 10 or GCC 11:

I'd expect this code to use the [division by a constant optimization](https://en.wikipedia.org/wiki/Division_algorithm#Division_by_a_constant), but it does not. [Godbolt link.](https://godbolt.org/#z:OYLghAFBqd5QCxAYwPYBMCmBRdBLAF1QCcAaPECAMzwBtMA7AQwFtMQByARg9KtQYEAysib0QXACx8BBAKoBnTAAUAHpwAMvAFYgATKVpMGoZAFI9AIQuXSS%2BsgJ4BlRugDCqWgFcWDfQZuADJ4DJgAcr4ARpjE%2BgAcBgAOqAqEzgyePn4SpClpTgIhYZEsMXF6AOwArHaYDoUMQgRMxARZvv56BvaYjhnNrQTFEdGxCT0tbR05XHZTw6GjZeNV1QCUdqjexMjsHBYAzKHIPlgA1GaH7sgKBOhYVFfYZhoAgq9vzGwKSUx750wBFonzMlWs73O5wA9NCAFQIxFI5Eo1Fo9EYzHoz5Q2HQnEw%2BFY4kk0lkpEE05MBQKc7IPC7bxGYgAfSi3ioVFibOpmAJYIhbyhSW8UVoeGQIFBkKhhOh5wAKghMOcCABPJIq7xKdDnfjEc5YAixFihULAVXK85pABeKtQVHOAEdvJhXQA6AlQ9Wax7WvB2lkEf2Bn18w6Cr2E8kx2OxqO2zBBjUqxMQdZ0gR3KMCqNQ4hAnYMc4QUKXQ4vQ4AEXO2wIGauADFzmWALS17zBkDnSxyRuN7AAJRZQgAkgAtbDndsQOvTlsMdZXQWyy6VKvSoWy2Fx3d7jFRqKoLyAlhJdXpzMMbMyqG52%2BygsEIsL8sb6sdgjLnPrzernf7oBQGHsetB6kytCXmg15fg%2B95bquT4vqWxY2OcXByA2ejVD2fYDsOY6Tm%2BVw1nW35wb%2B7xRgBQG0XGCYBkmYbnEwABuTB0EwYqYFBWawQha4rqu5xIcQxYsEwqgsmmGbtjJ5ECWCG5UQ%2BNF0eppIMaGKbnBJUkyVeN6KeCeYiYWYm4f2Q4jhOU7thhCmrkpf7bkSGnuViWlMTpoh/PSF4ZtBRlOSZD75uZxa9lZBG2fODkRj%2BykfDKSTEKgxqOJg6BSipAlqR5BUolG9KMsybIclyrJRLyECJsmmqWfhNmTiyS5hec3ZRU1hHYBAXXWT1rWmaQC4QBomHDZ%2BY0TRRQl3pRyUCUFwZ1cx/UxZOjmyqtOmhMuwkHXKVzuOco4MFgqiqqgloqmEqjBgA7sQhCYJ6D47Q1ZERod/6wsdp3nZgl1EDd5x3cGBZMOgb1bs5CW5bibmFcjyIEniaNIyjWMEsaZ5GMa5buGG3wqgqzyUkYNItiVyBMq05WcrEHXnCKYoSnSDK02V7KM1VvL8qFW6s%2BKkoud6KZ%2BgqP2yuxPjeZqW3i76mCOmTegAGzS2ZlWMHsiuqhLKuGcGaua7Ky0sgWOsMHr8MCWGktwlrKShMaxD6w7RvLYqTvm3xLIu4IsQKdRmNYwVofygA4kCLHG9rsS6yqIMEFaL0sCxwapyqVBpYItaOtn5w85VMO/fKbw0rEBC0sYgLEGlBp4IXVol0zeC0pgZ7qmXrnh9jD5W4nNs53nBDpj%2Bc2PhFLNt%2B71SWGR1TvlPzkI33/fI5H5wx8GTDx0PBYj1doPp5noO57IBeg3PveIxXVdtLXxaxI3LYtyqc8tp33dqnf0ab0KsVf2h8k56jHrxGCk9TKiWLEkOeZgF5LxXolMWADAEeW3rvOOoDj4pzTrjc%2BRdqrIAANbX2IRVWI/88TnErkoJ%2BLEX4NxIO/G%2BVCm4/3PH/UOGCI6DxVsPAEJDSET1mjAme8COGIMsGWEiJFzjjXLM2dazU7LoWZm2dCchEEoIoklXhfD3JYNjvvb2uCAT4JVGfJgWdW7/HIQ6dhvMaGwjoY/GuTD65v2bs4yq39TzcNcUY4xD4LYWM/g4yBwVZTwQOrA2e0iF5yOrAopRTZGoDVivZTRxZ7I6OXvrNei1fohI0iYveCcj6WOukXdOwSyl0SjBE2smpiC2NYTI3RtVGJBgXBdNqxkp7hWfBZKRvMZGzk7JcKw/SgZYRwqonquiikLUMY04Cqk3HYLMXxKpYCrEtkIbYm%2BDiKGtw4a4h%2BDDPF11fqw3xlDeYBK7kE9ZGz9zAJgpbQR1T7RtI6QaLpy8emhjmaoQKfFoHtQSeMyqky5xoVCAMmZiy8JZM2oUu2IUDFbI%2BZsvKbiRQKAQFcuhDxn5HK7ifJ5pdt6jg/sXDhASaC0HoLqbOxYi5eCwHcKlGcO61lYrEJ6hBjQMAafivcUZWKoDwLqYlCAIDhN%2BWA9OgyQrDI6iAMID0SwWHVnC4OySGC6IzAqUsuMlzYuEikmsKEZmWG0QszJG1KyRnaojc4jYSDAFMUeaZ3LaC8uDAIV6plhQIKQZ2XR7oAB%2BFrrVaqhIij8UzgxoQwi6pZtlVm4sJVKz5Wz5TKG1AgOOX9L4Zzri9dpRADTtJMOGvFhbdxRlxkkfGKpjrE1YKTUcbtAXkwfLK%2BVLMy0QAVAO2IgK9QMjuCNb2U7B31oNecSm9ZoUCShA9BAdAVTUHncGMAYAFEbo1cJOJh1FUQDhDQYg2YrA2CTaZYpOK0H5VbWSbeKQkjvK/d%2BkdcqFWoCSGIoZpkjXz0XjG5e8bE36xTdMhR6bHXOtRa6tReaP1hwA8SH9oGICLn/Xh/DQGx2/tBfLW6F75pat3fuksDBWzthPWkma27BIRpZoRl97U32xLWS20jmli3nHcPQVotJaXUJIyJg85HdSnEwK0cDmruMMfoCWE9ryAq0cvYLZ2vHEOCaSu%2BlKaUMrGmyjh%2BTonCUlss30azxtiDeEcCQSVdnMRRglJzOmVUOEQEDm7RJvMWQjQ%2BiqPSLJ6AmFTvpqE3YaYBYZpVHkSgIAxbi8AVOaGs2TSg8FueQ1xH6LQSFpmUGQ4WbwOxY0OUSkb28z5sTVZMB3Dcx54gXmWtogJKldKzmsqNdXHGvzpV6ZzzU4J1eC15p6Ka5%2BvrqIMYVznaoLKdImB%2BUIGqDmk2DS3zWwDN2zAwLHZlMtlbqMZQdq7YTXtbBFSRd6cGAAsm8AAGmolkw6txUipil7mzLuzC3ZhNrmU3pHXDJhWAWQlwei3XlCO4tj2beye9Y4H0OJmw%2BeCAEAUXzifZ%2Bz1N8BtNQk2pv5kHeP3Bw%2BwITqLEBSe/b401xGN3sRic8DBLr9bevc8RMVWnuPKozdXN2SHqWEH44rBAAsrtYipSBCyUQ2ZYdwmeBAOemwSffbUfpq9C3bPC6Krz0D%2B2%2Bedfc4LuTwvReHbS7EZVeyceBYmRrWs2diCJc0R7l37t5e9SV0HYgquCDq%2BpF%2BLXOu9cjTZz1Y3hnbWOggKnQVbGPwGvSsqP3r7U8HUzwoVszwb15%2BoTEYAoR0wjUrz1tw6YOcHQE6blH6Dzci7E/QvAwAGBsHzqBmdq7hNd4RE7qHnv4Xe%2BH3Wzp1Y3cwQO1PoPa6G8p61b4jPe7aTZ5rLn33/uTfCRL2XisynVMt8Omf8vE6G/umr7X/XD%2Bm/rGv%2BZzn09RnFjhCX7DuUDu3O287Wtu3WQu5uUYcage02m%2Bpkt%2BF%2BkmxAE8Nq7eTWqUdWti7AOG8oSoWoDAZoTgYgHcW2dwJATAfq/8UsUalgSesUmaKyNqcMHqHwQmbwsq%2BMjGrsLECko6uo1I6Al4J%2BQItAhOMBMO7grsI0XAeghwzwc6NoNouYv4qBccCiNAih7ol8ggKBU8mmPETA/uUIGhNo7olGH%2BahH4JhWhECHOSkHAmwohHA1QvA/gHAWgpAqAnA7gNgaECg2wuw3ashegvABAmgDhmwpCIA1Qhw7o6s6s8QlQXAlQGgkglQkg1QXACRhgnAkgrh4RnhnAvACgIAGgpAYR7hDhpAcAsASAaAZ4%2B6ZAFAbuDR9AcQrQLAXAGgegcwrKbsJRuuBRUQoQrQaonAPApA9Rg%2BBAAA8gwLQGMZUaQFgBJCYOIEsfgAWP0MKiUUsUDH0J2PsBMcrk4R4eKFEO0sQGqJ4FgAUQQM9CwOMVUVQEYMAAoAAGp4CYAPQzFU5PEyCCAiBiDsBSAAnyBKBqAFG6AGBGAmAoC%2BFWCGB4BRAlGQCbCgaNC7GtgzFUAx7TgsCpzRDFH1DOYZCuDnQzBdBlHBBLClDlD6BzD5DpACCUm5BMmNAjB0njBcC1C9D9ACCDDTBeCdD6BlF8mNCCmLAlBjAVBzBo5CnZD%2BBykLCckykSAbBbA7B7D6DlEFhHFVFOEuGkBuEeFeEcBvCDjvbnDADIDIDoQaDuh6DuhcBMZhoZgQA%2BFPqInnC4CECdLBEjSeCtFMxHB6DrChHhHv6kBRGSDqzuiHCHAyGJAACc1Q2Ekg8QCZORHAeRxpBRZpxRpR5RkZjhnAIReZSxBZxZlRUZwqD6ZJkgQAA%3D)

```
volatile int a;
void asd() {
    etl::circular_buffer<int, 123> fizz{};
    a = fizz.front();
    while(a)
        fizz.pop();
    a = fizz.front();
}
```

```
…
.L3:
        adds    r0, r1, #1
        movs    r1, #124
        bl      __aeabi_uidivmod
        ldr     r3, [r4]
        cmp     r3, #0
        bne     .L3
…
```

I'm unable to reproduce this on Godbolt, but I'm also not seeing the power of two division optimization happen with my code. For example, I'd expect `etl::circular_buffer<uint_fast8_t, 7> requestedChannels{}; requestedChannels.pop()` to convert down to `& 8`, but I'm seeing a call to divide in my disassembly.

I'm able to help the compiler out a bit here and get optimal code though:

```
        if(this->requestedChannels.capacity() != 7) {
            __builtin_unreachable();
        }
        this->requestedChannels.pop();
```

It seems to me like the compiler is having trouble understanding that `BUFFER_SIZE` is a compile-time constant in sufficiently-complicated code. Perhaps specialization of the function would help it out.

jwellbelove, 2022-03-26:
I played around with the Godbolt compiler settings.
It seems to be the presence of the `-mthumb` option that makes a big difference for GCC.
The call to `__aeabi_uidivmod` is also not present in GCC 10.3.0 on Godbolt with -`Ofast -mthumb`.
I'm not sure why this is different for your compiler on your PC.

jwellbelove, 2022-03-26:
The `__aeabi_uidivmod` also disappears if the body of `pop()` is replaced with

```
pbuffer[out].~T();
++out;
if (out == BUFFER_SIZE)
{
  out = 0U;
}
```

jwellbelove, 2022-03-26:
Adding `[[unlikely]]` results in slightly shorter assembler.
This is used in the ETL by using the macro `ETL_UNLIKELY`.

```
pbuffer[out].~T();
++out;
if (out == BUFFER_SIZE) ETL_UNLIKELY
{
  out = 0U;
}
```

flaviut, 2022-03-26:
Looks like that's also how `queue.h` is implemented. And it turns out that's just fine for my purposes anyway, I don't need the overwrite-on-wrap functionality of the circular queue now that I think about my code more.

Anyway, I don't know if any action needs to be taken here. I've always been told that compilers are great, but it's very surprising to see one miss such an obvious opportunity.

jwellbelove, 2022-03-26:
I may add the optimisations to the next release, or a minor fix release anyway.

closed 2022-03-26 by jwellbelove