ETLCPP/etl #519: cumulative moving average doesn't work the way it's implemented in this library

flaviut, 2022-03-27:
BLUF:

- I first learned of all this stuff today, so there will mistakes
- `etl::cumulative_moving_average` doesn't do what it says on the tin
- `etl::cumulative_moving_average` isn't particularly useful anyway, even if it worked correctly
- an actual moving average requires O(n) space
- many applications actually want a low-pass filter, aka EWMA, which requires O(1) space
- most practical digital filters are specializations of infinite impulse response filters

-----

`etl::cumulative_moving_average` and [the blog post](https://www.etlcpp.com/blog/2019/06/18/implementing-a-moving-average/) misunderstand how CMA works. I'm particularly unhappy with the wording in the wikipedia page, since movement implies dropping off old values, while that is not the case at all. I've edited the page.

Unfortunately, there is not a way to calculate an average of a window of elements without keeping all of those elements around. I'm not going to try and prove this with math because I'm not smart enough for that; instead I'm going to try and show a series of programs that do various things and hopefully make my point.

The following program implements [CA according to the Wikipedia page](https://en.wikipedia.org/wiki/Moving_average#Cumulative_average) (python, because I've had enough of C++ today):

```
def cma(arr):
  n = 0
  avg = 0.0
  for v in arr:
    avg = (v + (n * avg))/(n+1)
    n += 1
    print(avg)
```
```
>>> cma([1,2,3,4,5,6,7,8,9,10,11])
1.0
1.5
2.0
2.5
3.0
3.5
4.0
4.5
5.0
5.5
6.0
```

The following implements CMA the ETL way:

```
def etl_cma(arr, preload=0.0, n=10):
  avg = preload
  for v in arr:
    avg = (v + (n * avg))/(n+1)
    print(avg)
```
```
>>> etl_cma([1,2,3,4,5,6,7,8,9,10,11])
0.09090909090909091
0.2644628099173554
0.5131480090157776
0.8301345536507069
1.2092132305915517
1.6447393005377742
2.1315811823070674
2.665073802097334
3.2409761837248485
3.8554328942953164
4.504938994813924
```

Even with a preload, the result is very different:

```
>>> etl_cma([1,2,3,4,5,6,7,8,9,10,11], preload=5.5)
5.090909090909091
4.8099173553719
4.645379413974455
4.586708558158595
4.624280507416905
4.74934591583355
4.953950832575955
5.230864393250868
5.573513084773516
5.9759209861577425
6.432655441961585
```

-----

The operation that is intended here is likely [a simple moving average](https://en.wikipedia.org/wiki/Moving_average#Simple_moving_average), which requires storage of the last k data points.

A simple moving average is a specialization of [a infinite input response filter](https://en.wikipedia.org/wiki/Infinite_impulse_response), where b0…b(k-1)=1, a1…ak=0, and a0=k, and requires storage of k input values.

For my application, I want a single-pole low-pass filter (i.e. equivalent to a single RC filter). That's another specialization of a infinite input response filter, but only requiring storage of one value. [In this case](https://en.wikipedia.org/wiki/Low-pass_filter#Simple_infinite_impulse_response_filter), a0=1, a1=-(1-α), b0=α, where α=(2pi(1/samplingFreq)(cutoffFreq))/((2pi(1/samplingFreq)(cutoffFreq)) + 1).

flaviut, 2022-03-27:
Oh, and I improved my understanding of this while [trying to refactor the CMA code to be more general](https://github.com/ETLCPP/etl/compare/master...flaviut:moving-average). Unfortunately, as my understanding of the issue has grown deeper, my lack of C++ library design skills has become more evident :(

jwellbelove, 2022-03-27:
Yes, I realise now that I conflated had SMA and CMA in the design of `etl::cumulative_moving_average`. I took the 'n' in the Wikipedia article to be a fixed window size, rather than a continuously incrementing value.
The original intent of the class was to allow an SMA style average without the overhead of having to have N samples stored.
I already knew that it did not have exactly the same response as a real SMA, but the design aim was to allow simple smoothing of input data, over a window, with the _smallest_ possible overhead. It was never intended to implement any particular filter design. Maybe I should rename it to something that indicates its simplified design brief.

flaviut, 2022-03-27:
> Maybe I should rename it to something that indicates its simplified design brief.

I think a renaming & a short explanation of what it does would be the right approach here.

As far as usefulness goes, it looks like the -3db (and I could be misunderstanding everything) point is 1.2279471772 * SAMPLES samples in:

```
>>> etl_cma([10]*13)
…
7.103356202633122
>>> etl_cma([100000000]*122794717, n=100000000)
…
70710677.72499876
```

I can't talk about the slope of the filter past that, since I don't understand the math.

This is far better than my gut feeling of "this is taking way too long to settle" that I got from looking at my code & allows me to figure out what filtering function I actually want here.

closed 2022-06-24 by jwellbelove

doniks, 2024-01-05:
the documentation still talks about cumulative_moving_average: https://www.etlcpp.com/cumulative_moving_average.html

jwellbelove, 2024-01-05:
Although that web page still exists, it is not linked to by the menu.
I will remove it though.