ETLCPP/etl #521: CMake: Add package version file for installation

robamu, 2022-03-28:
Hi,

CMake provides the option to write a package version file. The documentation for this can be found [here](https://cmake.org/cmake/help/latest/module/CMakePackageConfigHelpers.html#command:write_basic_package_version_file).

I think this is really useful if `FetchContent` and `find_package` are used to find an installed ETL library. From what I have seen, I was not able to use the version feature, for example with 

```cmake
find_package(etl 20 QUIET)
```

Could be that I misused the feature or did something wrong so correct me if I am wrong. If I omit the version number, CMake is able to find the installed ETL library without issues.

Also I think it would be nice if installation instructions are provided for CMake.

I followed the following steps to install a specific version of ETL

```sh
https://github.com/ETLCPP/etl
cd etl
mkdir build && cd build
git checkout <someVersion>
cmake .. && cmake --install .
```

jwellbelove, 2022-03-28:
I will take a look, but I'm no expert in cmake.

robamu, 2022-03-29:
I hope this is not already something you have been working on, but I opened a PR which introduces a little bit of version handling to the CMake file:
https://github.com/ETLCPP/etl/pull/522

jwellbelove, 2022-04-09:
Added 20.27.2

closed 2022-04-09 by jwellbelove