ETLCPP/etl #544: `substr` of `etl::string_view` does not compile with C++11

janoschreinking, 2022-05-10:
Simple taking a substring (`view.substr(start, end)`) leads to an compile error with C++11 via GCC:

```cpp
error: assignment of member ‘etl::basic_string_view<char>::mbegin’ in read-only object
       mbegin = other.mbegin;
```

```cpp
    //*************************************************************************
    /// Assign from a view.
    //*************************************************************************
    ETL_CONSTEXPR etl::basic_string_view<T, TTraits>& operator=(const etl::basic_string_view<T, TTraits>& other)
    {
      mbegin = other.mbegin;
      mend   = other.mend;
      return *this;
    }
```
I guess that `ETL_CONSTEXPR` will be replaced by `constexpr` which makes the member function implicit `const` in C++11. According to the other function next to `operator=` it should be `ETL_CONSTEXPR14`. This code works fine:

```cpp
    //*************************************************************************
    /// Assign from a view.
    //*************************************************************************
    ETL_CONSTEXPR14 etl::basic_string_view<T, TTraits>& operator=(const etl::basic_string_view<T, TTraits>& other)
    {
      mbegin = other.mbegin;
      mend   = other.mend;
      return *this;
    }
```

jwellbelove, 2022-05-11:
Thanks, I'll fix that in the next release.

jwellbelove, 2022-05-19:
Fixed in 20.28.0

closed 2022-05-19 by jwellbelove