ETLCPP/etl #546: Unable to use bitset with constexpr

jputcu, 2022-05-20:
I wanted to use `etl::bitset` as `constexpr` but the compiler gives an error:
~~~~
error: the type 'const etl::bitset<10>' of 'constexpr' variable 'bs_etl' is not literal
~~~~

`std::bitset` does work.

As a demonstration, https://godbolt.org/z/5M74fYboo

jwellbelove, 2022-05-20:
I'm afraid that most of the containers in the ETL are not constexpr yet.
I need to put some time aside to gradually work through them.

jwellbelove, 2022-08-02:
I have discovered through experiment that the technique the ETL uses to reduce code bloat (by passing a pointer to the fixed sized storage to an implementation base class) means that any class that uses it cannot be `constexpr`.

The code below demonstrates the issue.

```
struct Base
{
  constexpr Base(char* p_)
    : p(p_)
  {

  }

  char* p;
};

struct Derived : public Base
{
  constexpr Derived()
    : Base(data)
    , data()
  {

  }

  char data[100];
};

constexpr Derived d;
```

```
error: ‘Derived{Base{((char*)(& d.Derived::data))}, char [100]{'\000', '\000'}}’ is not a constant expression
   67 |       constexpr Derived d;
      |                         ^
```

jputcu, 2022-08-02:
Too bad it doesn't work. If this technique is used all over ETL it means many lost optimizations.
I think the issue is that the base class must/will be constructed before array: chicken & egg

jwellbelove, 2022-08-02:
The problem with not doing it that way is that every different sized container would be unique, thus creating mass duplication of code.

i.e. If there was no `etl::ivector<T>` then 
```
etl::vector<int, 10> v1;
etl::vector<int, 20> v2;
etl::vector<int, 30> v3;
```
would generate three unique vector implementations for type `int`.

jputcu, 2022-08-02:
I have done it by passing the array pointer to methods of the base class.

jwellbelove, 2022-08-02:
Alas, doing it that way would be a breaking change.
`etl::ivector` would cease to exist as a reference type for all sized vectors.
Any functions that took an `etl::ivector` would need to be specifically sized, or become a template.

jputcu, 2022-08-03:
Not constexpr related, but more about the code bloat: some time ago I was going to replace my `uint8_t` with a `etl::bitset<8>`, but this added hundreds of bytes of (AVR) flash. I was expecting this all to be zero cost.

jwellbelove, 2022-08-03:
I tried with/without global `etl::bitset<8>` variable declaration, with Keil and ARM6 compiler and got these results.

Without bitset variable
-O0 Code=1580
-O1 Code=782
-O2 Code=898
-O3 Code=898

With bitset variable
-O0 Code=2404
-O1 Code=782
-O2 Code=898
-O2 Code=898

jputcu, 2022-08-03:
Quickly replace that `uint8_t` again with the `etl::bitset<8>`: went from  76354 bytes to 76582, only using the `test` and `set` action.

Flags:
~~~~
-Os -DNDEBUG -funsigned-char -mrelax -ffunction-sections -fdata-sections -mcall-prologues -fno-inline-small-functions -fno-rtti -fno-exceptions -fno-threadsafe-statics -DETL_TARGET_DEVICE_GENERIC -DETL_TARGET_OS_NONE -DETL_NO_STL -DETL_NO_CPP_NAN_SUPPORT -DETL_NO_HUGE_VAL_SUPPORT -DETL_USE_LEGACY_VARIANT  -g -std=c++1z
~~~~

jputcu, 2022-08-03:
I did a small test on compiler explorer: https://godbolt.org/z/TY71Ts1zT.
When using gcc 8.5 and blow, I get 63 lines for x86-64, selecting gcc 9.1 and up gives me 2 lines.

jputcu, 2022-08-03:
Using `std::bitset` always gives me 2 lines and can be `constexpr`.

jwellbelove, 2022-08-03:
If I add calls to `test` and `set` I get...

-O0 +344 bytes
-O1 +136 bytes
-O2 +24 bytes
-O3 +24 bytes

jwellbelove, 2022-08-03:
On a slightly separate note; How does Godbolt know where to get the ETL headers from? I can't seem to see any settings for that.

jputcu, 2022-08-04:
With the button "Libraries" in the disassembly view. Only the `trunk` version is available.

jwellbelove, 2022-09-12:
Fixed 20.34.0
New bitset implementation
https://www.etlcpp.com/bitset.html

closed 2022-09-12 by jwellbelove