ETLCPP/etl #547: Debounce States conflict with Arduino

lekoook, 2022-05-22:
One of enum for states in debounce.h is named "CHANGE".  The Arduino framework has a #define CHANGE used for their interrupts methods.

This appears to conflict with the Arduino framework when both are used together and compilation will fail. The Arduino library version also suffers from the same issue.

A simple solution may be to just rename this to "CHANGED".

jwellbelove, 2022-05-22:
It's a pain that Arduino seems to have created all of its constants a plain old fashioned #defines, as they do not recognise any scope. The ETL constants are wrapped in structs/enumerations.

The only real solution to stop this happening again and again is to modify the ETL enumerations to non all caps, such as `Change`.

This will of course be a breaking change.

lekoook, 2022-06-01:
> A person on the ETL Slack group suggested a workaround last year, by including `debounce.h` before `Arduino.h`
> 
> ```
> #include <etl/debounce.h>
> #include <Arduino.h>
> ```

Things can get rather complicated as the project grows, difficult to track the order of the #includes.

The temporary solution I've been using so far is to `#undef CHANGE` just before `#include <etl/debounce.h>` and `#define CHANGE 4` right after. Ugly but I can't think of another solution that isn't a breaking change.

jwellbelove, 2022-06-22:
Actually I've just realised, looking at the code, that the enumerations are purely internal to the debounce class, so I can name them how I like.

jwellbelove, 2022-06-24:
Fixed 20.29.2

closed 2022-06-24 by jwellbelove