ETLCPP/etl #548: Issues with ARCH_INDEPENDENT flag when cross-compiling

robamu, 2022-05-22:
I installed the library with the new `ARCH_INDEPENDENT` flag. This will cause CMake to also find an ETL package installed into the host root filesystem (x86-64 Ubuntu 22.04).

I am currently trying to cross-compile an application for a STM32H7 target running FreeRTOS. I use `find_package` to detect any installed ETL libraries and fall back to using `FetchContent` declare if none are detected.

CMake seems to be able to detect the ETL installation on the host. However, the `stm32-cmake` cross-compilation helper does not seems to be able to pass the right compilation flags or find the library om the host. It fails with an error, not being able to detect
the `<etl/multimap.h>` header I am using.

In the CMake toolchain file, the following configuration is used:

```cmake
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY BOTH)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE BOTH)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE BOTH)
```

I would've expected the BOTH settings to solve this issue, but the headers still can not be found and at  least the bare metal compiler does not appear to search the host paths.

When I remove `ARCH_INDEPENDENT` from the ETL installation, the issue is solved because CMake will fall back to use `FetchContent` . It will not accept the package, probably because it was installed for a different architecture. 

@g-arjones do you have an idea what the reason might be that I can not use the host installed ETL or why my cross-compiler is not able to detect or pass the include header paths to the build command?

g-arjones, 2022-05-23:
Could you please share your `CMakeLists.txt` file?

robamu, 2022-05-23:
You can find it here: https://egit.irs.uni-stuttgart.de/fsfw/fsfw-example-stm32h7-freertos/src/branch/master/CMakeLists.txt . 
I could try to provide a smaller MWE using some simpler stm32-cmake example..

g-arjones, 2022-05-24:
I can't reproduce the issue and I have no idea why cmake would find the package, and import the target but fail to add the include directory....

robamu, 2022-05-24:
Can you maybe post the resulting build command / added compiler flags when you are cross-compiling some application and trying to use the ETL installed on the host?

g-arjones, 2022-05-24:
One thing you can do to try to debug your setup is to edit the installed `etlConfig.cmake` file, and add a debug statement such as:

```cmake
message(STATUS "etl include dir: ${_IMPORT_PREFIX}")
```

Just before the `etl::etl` target property is set (look for `set_target_properties` in that file). That's where etl's headers should be.

robamu, 2022-05-28:
Thanks, I will have a look

robamu, 2022-05-30:
In both the case for regular compilation and cross-compilation, the resulting include path is `/include` :

`etl include dir: /include`

I also have a smaller example here now: https://github.com/robamu-org/stm32-cmake-projects

Which can be compiled with the following steps, provided a cross-compiler is installed

```sh
git submodule init
git submodule update STM32CubeH7
cd projects
mkdir build && cd build
cmake -DBUILD_H743ZI=ON ../etl
cmake --build . -j -v
```
With an installed ETL, I still have the issue that the library is found by `find_package` but the host include path is not passed to the build command.

g-arjones, 2022-05-30:
Did you confirm that etl headers are really installed in `/include` ? That's a very odd path.

robamu, 2022-05-30:
It is an odd path. The prefix variable also just appears to be empty. etl is installed into the usr/local directory, so the headers can be found inside usr/local/include/etl/ . Somehow, everything works for a regular host build though. What I have not tried yet is setting CMAKE_FIND_ROOT_PATH_MODE_PACKAGE to ONLY. That could at least prevent CMake from accepting a package not installed into the cross compiler sysroot..

g-arjones, 2022-05-30:
Ok. So you just found the issue and it has nothing to do with `ARCH_INDEPENDENT`. If the headers are installed in `/usr/local/include` and cmake is passing `/include` to the compiler it will not find the headers.

My bet is that you didn't set the prefix/install paths correctly when you built etl.

g-arjones, 2022-05-30:
This issue can be closed, IMO.

jwellbelove, 2022-05-30:
Glad you managed to sort it all out.

robamu, 2022-05-30:
Hmm, I tried `-DCMAKE_INSTALL_DATADIR=/usr/local` and it did not solve the issue. It will still use `/include` as the include directory. Maybe it is some other directive. I agree it has nothing to do with `ARCH_INDEPENDENT` though.

Setting `set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)` has the effect that at least CMake won't accept the package anymore if it was not installed into the cross-compiler sysroot as expected.

I also still find this kind of confusing. If CMake is able to find the package, I would also expect it to be able to correctly infer all the correct include paths without me having to specify special prefixes during build/install time.

g-arjones, 2022-05-30:
> It will still use /include as the include

You have to pass this when you install `etl` (you will have to rebuild it). not to tor package. If that still doesn't work, you can try setting `CMAKE_INSTALL_INCLUDEDIR=/usr/local/include` as well ([this is why](https://github.com/ETLCPP/etl/blob/master/CMakeLists.txt#L26)).

> If CMake is able to find the package with find_package, I would also expect it to be able to correctly infer all the correct include paths 

CMake doesn't infer anything, the include directory is hardcoded in the generated targets file (you saw that yourself through that debug statement).

If `CMAKE_INSTALL_PREFIX` and `CMAKE_INSTALL_INCLUDEDIR` don't match, the generated file will be broken.

g-arjones, 2022-05-30:
Also, make sure you remove the previous installation by doing an `rm -rf /etl`

robamu, 2022-05-30:
The `CMAKE_INSTALL_INCLUDEDIR` appears to have done the trick, thanks. I can also now see the `-isystem /usr/local/include` argument in the build commands now which I was expecting.

robamu, 2022-05-30:
`etl` (or at least the data/share files) was/is not installed in the root tree anymore. That was fixed as part of https://github.com/ETLCPP/etl/pull/539