ETLCPP/etl #549: Issues with etl::atomic and enum_type

opcod3, 2022-05-29:
This makes it impossible to wrap it into an etl::atomic.

This can be resolved by removing the copy constructor and assignment operator.

This results in the following definition of `enum_type`:
```c++
#define ETL_DECLARE_ENUM_TYPE(TypeName, ValueType) \
  typedef ValueType value_type; \
	ETL_CONSTEXPR TypeName() : value(static_cast<enum_type>(value_type())) {} \
	ETL_CONSTEXPR TypeName(enum_type value_) : value(value_) {} \
  ETL_CONSTEXPR explicit TypeName(value_type value_) : value(static_cast<enum_type>(value_)) {} \
	ETL_CONSTEXPR operator value_type() const {return static_cast<value_type>(value);} \
  ETL_CONSTEXPR value_type get_value() const {return static_cast<value_type>(value);} \
  ETL_CONSTEXPR enum_type get_enum() const {return value;} \
  ETL_CONSTEXPR14 const char* c_str() const \
  { \
    switch (value) \
    {
```

Once this is done etl::atomic throws an error that std::atomic does not:

```
atomic_std.h:61:15: error: invalid conversion from 'int' to 'next::MessageType::enum_type'
```

This again can be resolved by removing or replacing the default constructor with the following:
```c++
atomic() = default;
```

jwellbelove, 2022-05-29:
I will look at that.
I will have to create an alternate version for C++11 and above.

jwellbelove, 2022-05-30:
I've got `enum_type` to work OK with `std::atomic`, but I'm currently trying to figure out how to get it to work with the GCC/Clang + No STL version that uses the atomic intrinsics. They expect a pointer to an integral value, not a structure.

jwellbelove, 2022-06-01:
The design aims of the ETL does not include creating a 100% compatible clone of the STL. That would be a monumental amount of work.

I try to add things to the ETL that are generally useful to the embedded programmer; _sometimes_ this includes reverse engineering parts of the STL. Not all of the STL is easily cloned, as it may rely on compiler specific implicit functions that may or may not be supported on a particular version. The type traits you mentioned are a case in point.

As the ETL is only maintained by one person (me) and I fund my work on it out of my personal savings and in my spare time (a complicated feature may see me working on it for 20-30 hours in a week), some features from the STL may take time to be implemented, if at all.

That said, I _am_ looking at creating a specialisation for non-integrals that use a mutex to control access to the atomic values.

opcod3, 2022-06-01:
Apologies I absolutely understand that you are the only maintainer. Apologies if I might have sounded entitled, I have no expectation of you implementing this.

I misunderstood when you said:
>  They expect a pointer to an integral value, not a structure.

To mean that the GCC and Clang STL expect that, and therefore tried to clarify the fact that the STL just expects a trivially copyable type. 

In my mind supporting the same thing for atomic in the etl would just mean a few line of code but I just now realize you don't have the luxury of only working with the latest compilers and have to support old c++ standards making it a much greater amount of work.

I am immensely thankful for this project as it has been very helpful in being able to rely on a pretty feature complete STL in an embedded environment. I am looking into getting my current employer to support the project, but as you can imagine sometimes that is an uphill battle.

jwellbelove, 2022-06-24:
Fixed 20.29.2

closed 2022-06-24 by jwellbelove