ETLCPP/etl #550: etl::vector unnecessay copy in constructor

TG-SAG, 2022-06-02:
If  you construct the etl::vector container with count copies of elements with a specific value (see `vector(size_t initial_size, typename etl::ivector<T>::parameter_t value)`) I would expect that exactly count copies are made. This is the case for the std::vector but not for the etl::vector.

I use the following example (@see Godbolt: https://gcc.godbolt.org/z/EjqsTTz6K)

```CPP
#include <iostream>

#define USE_STL

#ifdef USE_STL
#include <vector>
#else
#include <etl/vector.h>
#endif

struct E
{
    E() { ++count; ++sum; std::puts("E()"); }
    ~E() { --count; std::puts("~E()"); }
    E(const E& other) { ++count; ++sum; std::puts("E(const E&)"); }
    E& operator=(const E& other) { std::puts("operator=()"); return *this; }
    static int count;
    static int sum;
};

int E::count = 0;
int E::sum = 0;

const size_t SIZE = 3;

#ifdef USE_STL
using Vec = std::vector<E>;
#else
using Vec = etl::vector<E, SIZE>;
#endif

int main()
{
    E e;
    Vec v(SIZE, e);

    std::cout << "count  = " << E::count << std::endl;
    std::cout << "sum    = " << E::sum << std::endl;
    std::cout << "v.size = " << v.size() << std::endl;
}

```

The ouput for STL is:

```
E()
E(const E&)
E(const E&)
E(const E&)
count  = 4
sum    = 4
v.size = 3
~E()
~E()
~E()
~E()
```

The ouput for ETL (please comment out USE_STL in the sources) is:

```
E()
E(const E&)
E(const E&)
E(const E&)
E(const E&)
~E()
count  = 4
sum    = 5
v.size = 3
~E()
~E()
~E()
~E()
```

As you can see there's one additinal copy in the ETL. This is due to the fact that the internel `resize(size_t new_size, T value)` Method has a by value parameter instead of an reference parameter.

Why is this a problem? Because someone might use a class that has side effects or is expensive to copy.

gerschdn, 2022-06-03:
I've made a fork that should fix the above problem: https://github.com/gerschdn/etl.git

gerschdn, 2022-06-04:
Was fixed by #551 (Thank you)

closed 2022-06-04 by jwellbelove