ETLCPP/etl #560: Unable to upcast legacy variant

jputcu, 2022-06-29:
Maybe I'm using it incorrect, but having a legacy variant with 3 derived classes, requesting an upcast results in the following error:
~~~~
/include/etl/private/variant_legacy.h:981:46: note:   no known conversion for argument 1 from 'const etl::aligned_storage<88, 8>::type' to 'uint_least8_t*' {aka 'unsigned char*'}
~~~~
It appears the etl::aligned_storage conversion operators aren't considered.

jwellbelove, 2022-06-29:
You appear to be trying to get a **_non-const_** pointer from a **_const_** type.

jputcu, 2022-06-29:
I just replace
~~~~
const TBase* operator()(uint_least8_t * p_data, uint_least8_t typeId) const
~~~~
with
~~~~
const TBase* operator()(uint_least8_t const* p_data, uint_least8_t typeId) const
~~~~
at line 981 in `variant_legacy.h` and now it is ok. So indeed const related.

jwellbelove, 2022-06-29:
Is your variant object `const`?
Can you show me the line of code that calls `upcast`?

jputcu, 2022-06-29:
`m_transactions` is a variant array, which is const in this call.
~~~~
constexpr Id GetId(Transaction const *t) const {
    if (m_transactions[0].is_valid() && (&m_transactions[0].template upcast<Transaction>() == t))
      return 1;
    else if (m_transactions[1].is_valid() && (&m_transactions[1].template upcast<Transaction>() == t))
      return 2;
    else
      return 0;
  }
~~~~

jputcu, 2022-06-29:
I've been able to add a test to reproduce this. The extra `const` fixes it.
I also found a related issue, at line 1018 in `variant_legacy.h` there is a typo, `const U2&` should be `const U2*`.

jwellbelove, 2022-07-03:
In trying to resolve a compatibility issue between compilers, I realised that I could simplify the code and completely remove the `upcast_functor` classes from legacy variant. I am also adding a couple of extra functions `upcast_ptr` and `is_base_of`.

`upcast_ptr` returns a pointer, cast to the base type or `nullptr` if the type is not a base. 
`upcast` returns a reference, cast to the base type or asserts an `etl::variant_not_a_base`.
`is_base_of` returns `true` if the specified type is a base of the current value type, otherwise `false`.

jputcu, 2022-07-03:
Sounds good, in my case I could use the `upcast_ptr` and so avoid the `is_valid` check.

jputcu, 2022-07-03:
I noticed that you've put the variant in the `etl_legacy`. I had to add `ETL_USE_LEGACY_VARIANT` define to use them now, can I leave that out and explicit use `etl_legacy::variant` or `etl::variant`?
I ask because I started with the legacy variant and noticed the new one is more standard compliant but after converting my code its size exploded so reverted back.
So I was thinking that having an embedded `etl::legacy` namespace would have been nice so I could try the new variant in part of my code besides the old one.

jwellbelove, 2022-07-03:
The `etl_legacy` namespace is only active in the unit tests, to avoid unit test link issues with two classes name `variant`.

```
#if defined(ETL_IN_UNIT_TEST)
  namespace etl_legacy
#else
  namespace etl
#endif
```

The variant header auto select the variant to use.
If you are using C++11 then you should have always needed to define `ETL_USE_LEGACY_VARIANT`.

```
#if !ETL_USING_CPP11 || ETL_USE_LEGACY_VARIANT
  #include "private/variant_legacy.h"
#else
  #include "private/variant_variadic.h"
#endif
```

jputcu, 2022-07-03:
I'm using c++17 so I could use both, so an embedded namespace would be nice when fazing out the legacy variant.
Having the `etl::legacy` namespace always available could avoid the need for the `ETL_IN_UNIT_TEST`.
You could even play with `inline namespace legacy` inside `etl` for c++11.

jwellbelove, 2022-07-04:
I've modified the headers so that legacy variant is in the `etl::legacy` namespace if `ETL_IN_UNIT_TEST` is not defined.
I will try to take a look at why the variadic variant uses a lot more resources when I get time.

jwellbelove, 2022-07-04:
Fixed 20.30.0

closed 2022-07-04 by jwellbelove