ETLCPP/etl #563: member initialisation order warning in legacy variant

jputcu, 2022-07-15:
include/etl/private/variant_legacy.h: In instantiation of 'etl::variant<T1, T2, T3, T4, T5, T6, T7, T8>::variant() [with T1 = const Sas::HandpayInfo*; T2 = const Sas::ProgressiveInfo*; T3 = etl::null_type<3>; T4 = etl::null_type<4>; T5 = etl::null_type<5>; T6 = etl::null_type<6>; T7 = etl::null_type<7>; T8 = etl::null_type<8>]':
include/etl/private/variant_legacy.h:877:17: warning: 'etl::variant<const Sas::HandpayInfo*, const Sas::ProgressiveInfo*>::type_id' will be initialized after [-Wreorder]
       type_id_t type_id;
                 ^~~~~~~
include/etl/private/variant_legacy.h:872:60: warning:   'etl::aligned_storage<2, 1>::type etl::variant<const Sas::HandpayInfo*, const Sas::ProgressiveInfo*>::data' [-Wreorder]
       typename etl::aligned_storage<SIZE, ALIGNMENT>::type data;
                                                            ^~~~
include/etl/private/variant_legacy.h:455:7: warning:   when initialized here [-Wreorder]
       variant()
       ^~~~~~~

jputcu, 2022-07-15:
I saw that default, there are not warning flags used when compiling the tests, e.g. for `gcc` I added `-Wall -Wextra` which showed the above warning, besided many others.

So, just for a quick test, I added the following to `test/CMakeLists.txt`
~~~~
add_library(unit_test ${UNITTEST_SOURCE_FILES})
+target_compile_options(unit_test PUBLIC -Wall -Wextra)
~~~~

jwellbelove, 2022-07-15:
I've tried turning on `-Wall -Wextra` but I noticed a couple of warnings that seem to be false positives.

In `etl::state_chart` it complains about the member function `const transition* const transition_table_end() const` saying that the initial `const` qualifier is ignored.

```
state_chart.h:811:5: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     const transition* const transition_table_end() const
     ^~~~~
```
This is incorrect as a return type of `transition* const` has a different effect to `const transition* const`.
The first will allow you to modify the value pointed to by the returned pointer, whilst the second will not.

gcc.exe (x86_64-posix-seh-rev0, Built by MinGW-W64 project) 8.1.0

jputcu, 2022-07-15:
You could put only that test in a separate test build and build the rest using the stricter warning flags.
On the other hand, I believe the warning is correct, the method is const and transition is a class, so the const is indeed redudant.

jputcu, 2022-07-15:
It is actually the second const that is redudant.

jwellbelove, 2022-07-15:
I get different behaviour with the two qualified return types.
If I remove the second `const` it allows the user to increment the returned pointer, which is not the desired behaviour.

```
struct StateChart
{
  struct Transition
  {
  };

  StateChart()
  {
  }

  const Transition* F1() const  // Returns a pointer to a const Transition'
  {
    return transitions;
  }

  const Transition* const F2() const  // Returns a const pointer to a const Transition
  {
    return transitions;
  }

  Transition transitions[10];
};

int main()
{
  StateChart s;

  const StateChart::Transition* pt1       = s.F1(); // Compiles.
  const StateChart::Transition* const pt2 = s.F2(); // Compiles.

  ++pt1; // Compiles, but is not the desired behaviour.
  ++pt2; // <<<< Compilation error. This is the desired behaviour.

  return 0;
}
```

jputcu, 2022-07-15:
You get a copy, which is in this case a pointer, you can do what you want with it. To avoid this you have to use some kind of proxy object that doesn't allow the increment.

jwellbelove, 2022-07-15:
Ah, I see.
I've just realised that` const StateChart::Transition* pt1 = s.F2();` also compiles fine, which is not what I originally desired either.

jwellbelove, 2022-07-25:
Fixed 20.31.3

closed 2022-07-25 by jwellbelove