ETLCPP/etl #571: Activate compiler warnings in tests

jputcu, 2022-07-24:
Enabling warnings on the unit tests would catch a number of issues before releasing, for example #563.

If it is OK, I would like to help introducing some stricter compiler checks and fixing the warnings that are generated?
I will do this in my own branch and afterwards creating a pull request.

jwellbelove, 2022-07-24:
I made a bit of a start on `feature/enable-gcc-Wall-and-Wextra-compiler-flags`.
Most of the warning in the unit tests seem to be with the CHECK_EQUAL macros with 'comparing signed and unsigned' warnings.

jputcu, 2022-07-25:
I see they are more related to `unittest++`, perhaps we can include the headers via CMake `target_include_directories` with `SYSTEM`, this hides compiler warnings for that directory.

jwellbelove, 2022-07-25:
I've tried to add `target_include_directories(etl_tests SYSTEM PRIVATE ${UNITTEST_DIR})` (and various combinations thereof) but it has no effect on the warnings in the unit test header `Checks.h`.

```
/mnt/d/Users/John/Programming/GitHub/etl/test/test_bit.cpp:204:7:   required from here
/mnt/d/Users/John/Programming/GitHub/etl/test/UnitTest++/Checks.h:21:22: warning: comparison of integer expressions of different signedness: ‘const int’ and ‘const unsigned int’ [-Wsign-compare]
   21 |       if (!(expected == actual))
```

jwellbelove, 2022-07-25:
I've currently got rid of the `Check.h` warnings by adding #pragma diagnostic push/pop into the header.

jwellbelove, 2022-07-28:
Fixed 20.32.0

closed 2022-07-28 by jwellbelove