ETLCPP/etl #572: Use standard feature testing

benedekkupper, 2022-07-24:
ETL is doing a lot of its own detection on which language features are supported. The problem is, it's not very accurate. I got here because by updating ETL in my [C++20 project](https://github.com/benedekkupper/stm32-i2c-hid) the build has failed. The problem boils down to the `#define ETL_NO_SMALL_CHAR_SUPPORT ETL_CPP20_NOT_SUPPORTED` line. This condition is correct as it is spelled out. However, GCC 10 sets `__cplusplus` to `201709L` for `-std=gnu++20`, so the macro gets an incorrect value, compilation fails due to duplicate definition of `char8_t`.

A better way to deal with these issues (at least from C++20) is relying on the [C++ standard feature tests](https://en.cppreference.com/w/cpp/feature_test). These same macros could be defined by ETL itself for earlier C++ versions, as long as the `__cplusplus` numbering pitfall mentioned above is handled well. (A big oversight from the feature test design is not adding a macro to check if feature test macros are available at all or not.)

benedekkupper, 2022-12-12:
Any timeline for merging this branch? The later it gets done, the more additional work it generates.

jwellbelove, 2022-12-13:
Between holidaying in Malta and catch a bad cold, I've not looked at the ETL for a few weeks. I have several things to work on for the ETL, so I'm going to have to try to work through them over xmas as I get the free time.

rolandreichweinbmw, 2026-09-23:
Can this issue be considered as done now?

ETL is currently already heavily using feature test macros.

benedekkupper, 2026-09-23:
I'm sad to see the work put in the PR wasted, but realistically the effort to rename the macros to their specific feature names is not worth it, which is why this didn't go anywhere. I guess changes will be made on a case-by-case basis depending on what breaks.

It would have been beneficial if this approach was at least adopted for new code additions, especially for recent language and standard library features, as using a cutting-edge compiler version sometimes means the feature set is in between two standard releases, and the current way of using either all or nothing from a standard release is suboptimal.

closed 2026-09-23 by benedekkupper