ETLCPP/etl #579: Unexpected namespace for variant with non-CPP11 compiler

Un-done, 2022-08-18:
When using variant on an older compiler (no CPP11) I expected to use it as `etl::variant` instead of `etl::legacy::variant` 

variant.h suggests that this is the desired behavior.
```
#if !ETL_USING_CPP11 || defined(ETL_USE_LEGACY_VARIANT)
  #include "private/variant_legacy.h"   // etl::variant
#else
  #include "private/variant_legacy.h"   // etl::legacy::variant
  #include "private/variant_variadic.h" // etl::variant
#endif
```
In variant_legacy.h the adding of the legacy namespace is dependent  on `ETL_USE_LEGACY_VARIANT` therefore it is added if only `ETL_USING_CPP11` is `false`

jwellbelove, 2022-08-21:
I'm changing the logic to this.

```
C++11 | Use Legacy | Legacy Namespace
  0   |     0      |   0
  0   |     1      |   0
  1   |     0      |   1
  1   |     1      |   0
```

Un-done, 2022-08-22:
Makes sense to me. Thank you for your work.

jwellbelove, 2022-08-24:
Fixed 20.33.0

closed 2022-08-24 by jwellbelove