ETLCPP/etl #582: `unordered_set` equality comparison fails when hash collisions occur

jesseli2002, 2022-08-26:
Minimum reproducible example - using a bad hash function:
```c++
#include <iostream>
#include <cstdint>

#include <etl/unordered_set.h>
#include <unordered_set>
#include <etl/unordered_map.h>
#include <unordered_map>

struct bad_hash
{
    // always has hash collisions
    std::size_t operator()(int) const { return 0; }
};

int main()
{
    {
        etl::unordered_set<int, 10, 10, bad_hash> set1;
        etl::unordered_set<int, 10, 10, bad_hash> set2;
        set1.insert(3);
        set1.insert(4);
        set2.insert(4);
        set2.insert(3);

        std::cout << "etl set equality: " << (set1 == set2) << std::endl;
    }
    {
        std::unordered_set<int, bad_hash> set1;
        std::unordered_set<int, bad_hash> set2;
        set1.insert(3);
        set1.insert(4);
        set2.insert(4);
        set2.insert(3);

        std::cout << "std set equality: " << (set1 == set2) << std::endl;
    }
}
```
Output (compiled with MSVC on Windows):
```
etl set equality: 0
std set equality: 1
```
I suspect this will happen with the other unordered containers too. Culprit is probably that `etl::equal` is used to compare for equality, which iterates through the container - but order of elements is undefined; within a bucket if the elements were inserted in different order then the bucket elements will be out of order. Notably [cppreference](https://en.cppreference.com/w/cpp/algorithm/equal) specifically warns about using `std::equal` to compare elements. 

Fix will probably need to manually iterate through buckets, with bucket sizes being compared and checking that each element from the left-hand-side is present in the right-hand-side bucket.

jesseli2002, 2022-08-26:
Worth noting that standard library specifies that equality comparison between two different unordered_sets can (should?) be done using `Key::operator==` rather than the provided `key_equal` function, and that behaviour is undefined if `operator==` behaves differently from `key_equal`

jwellbelove, 2022-08-26:
I've started to look at this, but I'm away in Wales for the bank holiday weekend. I'll be able to look at this again on Tuesday, or Wednesday (depending on when I feel like coming home)

closed 2022-09-12 by jwellbelove