ETLCPP/etl #586: Build warnings in message_packet.h in IAR (version 8.42.2)

jeremyerdmann, 2022-08-30:
The change de5a90a91c1c14434e26c52a3a7ebc7df7abba7a committed on 12/21/21 introduced compiler warnings for me when using the message_packet.  My environment is IAR (version 8.42.2).

My etl_profile.h contains the following:
```c++
#define ETL_LOG_ERRORS
#define ETL_CHECK_PUSH_POP
#define ETL_VERBOSE_ERRORS

#define ETL_CPP11_SUPPORTED 1
#define ETL_CPP14_SUPPORTED 1
#define ETL_CPP17_SUPPORTED 0
#define ETL_CPP20_SUPPORTED 0
#define ETL_ENDIAN_NATIVE 0
```

The following warnings are generated with the latest rev of the message_packet.h file from my application in IAR: cd18cc6af0bf282bced48da14b22255f1f77f2be on July 27th, 2022:

```text
\etl\include\etl\message_packet.h line 663
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, T7, T8, T9, T10, T11, T12, T13, T14, T15, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 945 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, T7, T8, T9, T10, T11, T12, T13, T14, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 1225
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, T7, T8, T9, T10, T11, T12, T13, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 1502 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, T7, T8, T9, T10, T11, T12, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 1774 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, T7, T8, T9, T10, T11, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 2044 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, T7, T8, T9, T10, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 2312 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, T7, T8, T9, void, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 2577 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, T7, T8, void, void, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 2837 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, T7, void, void, void, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 3095 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, T6, void, void, void, void, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 3351 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, T5, void, void, void, void, void, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 3604 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, T4, void, void, void, void, void, void, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 3852 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, T3, void, void, void, void, void, void, void, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 4098 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, T2, void, void, void, void, void, void, void, void, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"

\etl\include\etl\message_packet.h line 4342 
Warning[Pe445]: template parameter "TMessage" is not used in declaring the parameter types of function template "etl::message_packet<T1, void, void, void, void, void, void, void, void, void, void, void, void, void, void, void>::message_packet<TMessage,<unnamed>>(etl::imessage &&)"
```

I have found a few ways to get the warnings to go away, but it's not clear to me which if any might be most correct.  

Option 1: define `ETL_MESSAGE_PACKET_FORCE_CPP03_IMPLEMENTATION`
Option 2: on the lines immediately following the lines in question, change from `explicit message_packet(etl::imessage&& msg)` to `explicit message_packet(const TMessage& msg)` or `explicit message_packet(const TMessage& /*msg*/)`

This is a bit complicated/beyond my expertise, so I'm looking for advice on how best to proceed.  Option 1 seems the least intrusive, but feels like the wrong answer for the etl project if this is something others are likely to encounter as well.  I'd make the change in a branch and figure out how to put up a PR if Option 2 or some other similar change seems like the correct path.  I have compiled that locally in VS2019 and the unit tests appear to be passing in that environment.

Thanks in advance for advice on how to proceed.

-Jeremy

jwellbelove, 2022-08-31:
Can you try changing the function signature at line 666 (and the other equivalent lines, 948, 1228 etc) to read `explicit message_packet(TMessage&& msg)`

jwellbelove, 2022-08-31:
A function signature like 
```
template <typename T>
void f(T&& value)
```
can be a bit confusing as it is not an `rvalue` reference parameter function, but a _universal reference_ parameter function.

[](https://isocpp.org/blog/2012/11/universal-references-in-c11-scott-meyers#:~:text=If%20the%20expression%20initializing%20a%20universal%20reference%20is,you%20are%20able%20to%20distinguish%20lvalues%20from%20rvalues.)

jeremyerdmann, 2022-08-31:
John, that change does indeed resolve the issue.  Thanks for the reference on universal references (and other similar concepts).  I started to read, but quickly realized it's a heavy topic... perhaps this Labor Day weekend while at the campground campfire.

jwellbelove, 2022-08-31:
I was away for August bank holiday at the caving club HQ last weekend in the Brecon Beacons, in Wales. I tried not to think about programming and just enjoy the scenery. (Didn't always work though, but the laptop stayed at home)

jwellbelove, 2022-09-12:
Fixed 20.34.0

closed 2022-09-12 by jwellbelove