ETLCPP/etl #591: Automatically define __STDC_LIMIT_MACROS for appropriate compilers

dhebbeker, 2022-09-06:
According to [the manual for using ETL with different compilers](https://www.etlcpp.com/compilers.html) for some compilers it is required to have `__STDC_LIMIT_MACROS` defined. I wonder if it would be an improvement if that macro would be defined automatically by `platform.h` or its dependencies in case such a compiler is detected and the appropriate limit macros are not already defined?

dhebbeker, 2022-09-06:
Yes, I am compiling for Extended [Embedded C++](http://www.caravan.net/ec2plus/language.html), which is a derivative from C++98.

dhebbeker, 2022-09-09:
737b0bf6f715941b088c41554d99ed8ff4a50619 is not sufficient to resolve this issue, as often `stdint.h` is included *before* `platform.h`.

jwellbelove, 2022-09-09:
Thanks, I'll look at that some more.

jwellbelove, 2022-09-12:
Fixed 20.34.0

closed 2022-09-12 by jwellbelove