ETLCPP/etl #596: Feature request: Helper functions to convert strings to numeric values

oresk, 2022-09-10:
We have the to_string() functions that are really ergonomic, especially when paired with format. 

Having a utility that would do a conversion the other way round would be really usefull addition to ETL.

jwellbelove, 2022-09-11:
Do you think that the string passed should only contain the numeric string to convert, or should it search for the start and end of the numeric?

i.e. find non-numeric characters.
```
"abc# 123.45*sne"
      ^     ^
```

jesseli2002, 2022-09-13:
Something like [stoi](https://en.cppreference.com/w/cpp/string/basic_string/stol) in terms of API, perhaps?

jwellbelove, 2022-09-14:
Including conversion of strings in binary, octal & hex formats?
It may need an explicit base parameter or deduction from the prefix.

```
etl::to_arithmetic<int>("1234", etl::hex());
etl::to_arithmetic<int>("0x1234");
```

RallyTronics, 2022-09-14:
does anyone still use octal?

jwellbelove, 2022-09-16:
I'm sure someone does somewhere  :-)

jwellbelove, 2022-09-16:
- The return type of the functions will be an `etl::optional<T>`, where `T` is an arithmetic type.
- A failure to convert will result in a default constructed `etl::optional<T>`.
- I'm thinking that I should not bother handling stripping off any `0x`, `0X`, `0b` or `0B` prefix from hex and binary strings.
- I'm also thinking that binary, octal and hex strings that are prefixed with `-` are not converted (Or ignored?).
- An attempt to convert a negative decimal string to an unsigned type will also be a failure.
- A `+` prefix to any numeric string will be ignored.
- Whitespace will be ignored.
- Any failure to convert will be submitted to the error handler or thrown as an exception (dependent on configuration)

jwellbelove, 2022-09-16:
Examples:
`" +123  "` as signed decimal OK
`" -123  "` as signed decimal OK
`" -123  "` as unsigned decimal FAILURE
`" +123  "` as unsigned hex OK
`" 123  "` as signed hex FAILURE
`" -123  "` as signed hex FAILURE
`" 0x123  "` as unsigned hex FAILURE

jwellbelove, 2022-09-20:
At the moment, `""` is not an error (and returns a default constructed `etl::optional<T>`), but `"-"` or `"+"` _are_ errors.

oresk, 2022-09-20:
Hm, what would then be a difference between an error and a default constructed `etl::optional<t>`? 
From 4 comments above:

- A failure to convert will result in a default constructed etl::optional<T>.
- Any failure to convert will be submitted to the error handler or thrown as an exception (dependent on configuration)

I was under the impression that a default constructed `etl::optional<T>` indicated an error, and depending on the additional config it can throw an exception. 

Which one it is I would say an empty string would should have the same consequence as any other string that can not be converted.

jwellbelove, 2022-09-20:
Sorry for my confusing description.
When I said 'errors' I meant a call to `ETL_ASSERT`. A return of a default constructed `etl::optional<T>` is of course a '_failure to convert_' error, so I agree that an empty string should be treated the same as any other '`failure to convert`' error.

jwellbelove, 2022-09-22:
How important is the ability to have a `constexpr` conversion?
At the moment I am returning `etl::optional<T>`. Unfortunately, `etl::optional` cannot be `constexpr` due to the aligned storage member.

jwellbelove, 2022-09-24:
Floating point in radixes other than decimal?

jwellbelove, 2022-11-02:
Added in 20.35.0
[https://www.etlcpp.com/to_arithmetic.html](https://www.etlcpp.com/to_arithmetic.html)

closed 2022-11-02 by jwellbelove

oresk, 2022-11-03:
Perfect!

Will be using it over the weekend